portapack-mayhem/mayhem-firmware #796: RadioSonde

4x5HF, 2023-01-29:
----
Portapack H2+R1
after firmware upgrade from ver1.2 to 1.6 the radiosonde app stoped working!
The screen is blanked and the only way to return to home screen is reset.

gullradriel, 2023-03-27:
Any news on that part?

helmarw, 2023-04-04:
i have a different issue. I also just upgraded from 1.2 to 1.6

as soon as i start RadioSonde something next to the right button get very hot, to the verge i cant touch it.
luckily i realized this before it started smoking
[image]

i assume its either the CS8122 or the AK4951
to get rid of the heating ive to power down the device, its not enough to just leave the RadioSonde Menu

gullradriel, 2023-04-26:
Looks like a problem with one/both of your boards, and not radiosonde related anymore.
If I were you I would try to open a discussion about it with the seller itself.

closed 2023-04-26 by gullradriel

helmarw, 2023-04-27:
why do you think that? Why does it only happen with Radiosonde? can you explain?!
Is Radiosonde using one of the chips that get hot?!
I can use the headphone jack with no problem e.g. when listening to FM, so i dont know what gives you the idea...
I have this board since at least three years, and never had that kind of problem.
And of course its out of warranty by now ...