portapack-mayhem/mayhem-firmware #797: About the new function: amateur radio

wzsySRC, 2023-01-31:
Hello, I am a loyal user of Mayhem firmware and an amateur radio enthusiast. In my country, the use of radio equipment requires detection and registration. I want to use Hackrf as an amateur radio for inspection and registration, and use it legally and legally. My country requires that the initial frequencies be 144MHz-148MHz and 430.00MHz-440.00MHz. I see that there is a Microphone Transceiver function in Mayhem firmware. Is it possible to add frequency restrictions on the basis of this function to generate a new function for radio lovers to check and register - amateur radio. I look forward to your reply.

eried, 2023-01-31:
yeah it is good idea. 

You could thinker around how it will work (some non intrusive way to limit) and play with the code. The best way to get it implemented would be to start by yourself: add new checkbox in options with a country selector and limit the microphone app based on that

zackattack710, 2023-01-31:
Hello, my experience with amateur radio is limited. I will look into your
question further and see if someone else has done this with the microphone
transceiver function. I apologize for not being able to help more and look
forward to receiving my portapack so that I can learn more. I think
its cool you reached out to me from across the world.

TechJeeper, 2023-02-03:
As a Ham I really like this idea.

wzsySRC, 2023-02-06:
> 是的，这是个好主意。
> 
> 您可以考虑它是如何工作的（一些非侵入性的限制方式）并使用代码。实现它的最好方法是自己开始：在带有国家/地区选择器的选项中添加新复选框，并基于此限制麦克风应用程序

wzsySRC, 2023-02-06:
> yeah it is good idea.
> 
> You could thinker around how it will work (some non intrusive way to limit) and play with the code. The best way to get it implemented would be to start by yourself: add new checkbox in options with a country selector and limit the microphone app based on that

I'm not very good at writing code, but I'm willing to try it. However, I hope the author can provide this new function in the new version, because it is a very important thing for amateur radio lovers in our country.

gullradriel, 2023-03-21:
Well. It would be simpler to limit all the apps to some specific ranges than only one, because one can do this in radio model for all the apps.
The code itself can be pretty quick to generate, but doing a GUI for it is a pain.
I propose a different approach, based on a config file, in which you specify white lists, something like 'radio_frequency_whitelist.cfg' at root of sdcard. We could even have both a whitelist and a blacklist, but I think a whitelist would suffice. 
If the file exists, it's read at each retune. if a retune is outside the white list, it's cancelled (you stay on previous freq)
syntax would be as simple as 'a=startfreq,b=endfreq'.
If we limit the number of ranges, we can eventually keep them in memory and prevent a file open /read/close at each retune.
That's how I would tackle it, since it's not something you change.

qwqoffice, 2024-03-07:
hackrf one应该是没有型号核准码的

gullradriel, 2024-03-07:
@qwqoffice there is no code for a 'brand' or 'type'
Here we are talking about providing a support for frequency limitations, but it would be up to the user to set them.