portapack-mayhem/mayhem-firmware #798: Preamp is not working

dwadi, 2023-02-02:
Hi,
I am looking for information about preamp problem, I've just bought HackRF with PortaPack, and noticed that AMP turned to 1 it not doing anything. I am not sure if it was working before on old firmware that I've got.
Please give me any informations about fixing this issue.

Radiomix2000, 2023-02-02:
One of possible reasons - you have RX preamp burned out, possibly by electrostatic. You should replace it with new one. The same analog doesn't exist anymore (no production) and as I know there is new alternative, more sensitive and more electrostatic tolerant.

dwadi, 2023-02-02:
Which one should I replace then?

Radiomix2000, 2023-02-02:
MGA81563 should be replaced, RX one. There are 2 of them. One is RX and the second is TX. Cause you have issues with receive performance, you change RX one. You can also check, if TX amp is ok. Just set hackrf to transmit CW carrier, and try to change AMP from 1 to 0 and vise versa. And check with some kind of receiver, is the signal level changing.
MGA81563 is out of life cycle. I just checked, and you should order MGA82563. It's modern and better.
https://www.google.com/search?q=MGA82563

dwadi, 2023-02-02:
I've tested TX and it working OK

dwadi, 2023-02-02:
Is it OK? https://www.rf-microwave.com/en/agilent-technologies/mga-82563/gaas-mmic-amplifier-sot-363/mga-82563/ ?

Radiomix2000, 2023-02-02:
Yes, that's it. Be careful, when solder it. It's pretty small

dwadi, 2023-02-02:
SMD soldering is beyond me, I've found one guy from my country and he will do it for me

Commander-Crash, 2023-02-23:
You can try just getting these cheap LNA's from ali express or ebay 
[image]

gullradriel, 2023-03-21:
No an issue with Mayhem. Closing.