portapack-mayhem/mayhem-firmware #803: H1 older-pack upgrade issue on 1.6.0

skandragon, 2023-02-08:
**Describe the issue**

I cannot flash the latest (1.6.0) to my PortaPack H1.

**To Reproduce**
Steps to reproduce the behavior:

1.  Flash the "for the flash card" stuff to the SD card.
2. Flash the new firmware from the file `portapack_h1_h2-mayhem.bin` using `hackrf_spiflash`
3. Reboot the device by unplugging and re-plugging.
4. Observe the black screen that briefly flashes white once.
5. Try rebooting while holding down the up, down, right, and left, as well as center.
6. "down" causes the black screen to become white.
7. All others (up, right, left, and center) cause the black screen.

I have tried flashing the original mayhem firmware from the sharedbrained repo, and it works, but of course has no features that are interesting.

**My Hardware**

I have a H1, probably R1 but it may be R2.  As I understand things, the only difference is the audio chip, not the LCD driver, so I should at least see something I'd expect.

**Affected versions**

1.6.0.

Note that the latest nightly (`2023-02-08`) works if I boot with the UP button pressed at startup.

**Were you able to update the firmware before?**

Several years ago I have flashed it, but I understand DFU mode flashing (and had to use it to restore to a mostly working HackRF One mode) as well as the SPI flashing tools.

**Can you try the upgrade with a different PC/Portapack/HackRF?**

Tried different cables (although I'm shocked there isn't a firmware or transmission verification done on the data!  Perhaps it's cheap voltage-dropping cables?)

skandragon, 2023-02-08:
It also appears that I must hold down the DOWN button at each boot of the nightly, and even then it takes several tries before it will show the logo.  Once that happens, it works as expected.

gullradriel, 2023-02-10:
Hello.
If the latest stable (master) does not work for you but the latest nightly is, it just means that you'll have to use the nightly until we release a stable/master from it. There is really no harm doing so.
For the fact that you have to keep pressing the boot key upon reboots/restarts, it means that you do not have the battery button installed or it is dead and the persistent settings are not saved. Having a SDCARD installed may help some other apps preserve their settings too.

skandragon, 2023-02-13:
Ahh, thanks.  I'll check the battery.

Note that pressing the button doesn't always work; I often have to reset multiple times to get it to come up.  I'll see if the battery is causing that issue too.

closed 2023-03-21 by gullradriel