portapack-mayhem/mayhem-firmware #807: hope to add RSSI

gancaokuai, 2023-02-20:
I want to use hackrf for radio direction finding, and hope to achieve this by adding RSSI function

eried, 2023-02-20:
you could do it now but visually :) but I agree, it would be a simple app for helping to do this

gullradriel, 2023-02-20:
Some apps are implementing a visual RSSI, like the mic app and the the recon/scanner app.
Recon is displaying the bar graph and values of min/med/max RSSI.
What is the idea of the app ? Just a level meter with maybe an histogram of last X values ?

gancaokuai, 2023-02-20:
[1.md](https://github.com/eried/portapack-mayhem/files/10783272/1.md)

KimIV, 2023-02-20:
> I want to use hackrf for radio direction finding, and hope to achieve this by adding RSSI function
Can't a waterfall help you?

eried, 2023-02-20:
> What is the idea of the app ? Just a level meter with maybe an histogram of last X values ?

Yeah, I think that is what @gancaokuai suggests. Maybe level and audio beeper depending on intensity. Not sure about histogram.

gancaokuai, 2023-02-20:
@KimIV use the waterfall is to difficult to find the station

> > 我想使用hackrf进行无线电测向，并希望通过添加RSSI功能来实现这一点
> 
> 瀑布帮不了你吗？

use the waterfall is too difficult to find the station

gullradriel, 2023-03-10:
#827 is on the way :-)

gullradriel, 2023-03-21:
Done.

closed 2023-03-21 by gullradriel