portapack-mayhem/mayhem-firmware #809: Looking glass is missing signals in the first 20 Mhz

tsuijten, 2023-02-24:
**Describe the bug**
When I scan on frequency ranges the first 20Mhz of the spectrum always seem to be "empty" while there are definitely signals.

**To Reproduce**
Steps to reproduce the behavior:
1. Make two presets:
`2400,2484,WIFI_BT1`
`2380,2484,WIFI_BT2`
3. Go to `Looging Glass`
4. Mak sure you are in range of WiFi networks that transmit signals between 2400-2484 MHz.
5. Select `WIFI_BT1` preset and take a look at the section between 2400-2420 MHz: there don't seem to be any signals.
6. Select `WIFI_BT2` preset and take a look at the section between 2400-2420 MHz: signals are clearly present.

This behaviour is not only present in the 2400-2484 MHz range but all frequency ranges I have tested.

**Expected behavior**
With this preset `2400,2484,WIFI_BT1` I also expect to see the signals in the 2400-2420 MHz range

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 2023-02-23
* Latest Nightly release: 1.6.0

**Additional**
There seem to be no signals between 2400-2420 MHz
[image]

But there actually are if I increase the start frequency range:
[image]

euquiq, 2023-02-25:
Thanks for noticing this. I'm confirming this on my device. Let me take a look into the code and see if I missed something somewhere.

euquiq, 2023-02-26:
I think I got it ... 
[image]

closed 2023-03-02 by gullradriel