portapack-mayhem/mayhem-firmware #810: Show multiple aircraft on ADS-B Map

heurist1, 2023-02-26:
I have to be honest I have already developed most this feature, and so I'm describing things that I have a solution for.

**The main objective is to provide an ability to see other aircraft on the map, as well as the selected aircraft.**  
This shouldn't change the current functionality, in terms of how the item is selected or displayed.

I have looked into reducing the used CPU by the ADS-B receiver.  The last major change I made was a re-write of the decoder, which allowed the detection of many more aircraft.  Since then a small fix was performed on the scaling of the baseband samples received.  Which both of these changes my portapack ran out of CPU when a lot of aircraft were visible,  When I connect my receiver to  good antenna during the day, I see more than a screen full of active (with location) aircraft (see included screen shot).

I have looked at the functions that may be using a lot of CPU and made a few more efficient, however I don't know how to profile the full processing chain.  I have also split the way that the list of aircraft is updated and displayed to make it do half the work on alternate seconds, when there are are a significant number of entries.

Changes to the map include a list of other markers, and an ability to text with a transparent background rather than black.

Screen shot of list during the day (this used to result in a unresponsive GUI)
[image]

Proposed look to the map.
[image]

I will submit a pull request for the code within the next week.

Brumi-2021, 2023-02-26:
Congratulations for sharing your new improvement ideas !   
It seems very interesting feature.! 
We are Looking for testing it soon ! 

Recently I made two minor  PR‘s about ADSB TX (about bearing angle GUI Jump  , and adding VERT speed field  ). Now working about solving  #733 . Soon I also hope to release it’s PR.  
cheers

heurist1, 2023-03-02:
Pull Request #815 has been merged which adds in the feature described in this issue.
The issue can now be closed.

closed 2023-03-02 by heurist1