portapack-mayhem/mayhem-firmware #813: APRS RX  FW 1.6.0 / nightly 230223 - not decoding

FrankUlbrich, 2023-02-28:
**Describe the bug**
Hello group,
can anyone confirm that APRS RX with 1200Baud is working in Germany on 144.800 MHz with the firmware below?
I tried APRS RX with both versions listed below connected to my 2m groudplane antenna on the roof of my house.
It is possibel to hear the NF sound when the packages are received with my hack rf one with PortPack H2+,
but no data/packages is decoded in the "stream"/"list" tab of the APRS RX App.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Receive'
2. Tap on 'APRS'
3. Switch from NA to EUR - 144.8000 is now displayed
4. Settings are "144.8 MHz, 0, 40, 40
5. NF sound is clear to hear in the headphone, (rec) not pressed

**Expected behavior**
Data shown on "List" or "Stream".

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:   1.6.0
* Latest Nightly release:  230223
* Previous working release: unknown

**Additional**
My device was preloaded with 1.6.0 I updated to the latest nightly.
Maybe I missed something to do in the "APRS RX" App.
Help is really appreciated or the confirmation that decoding works with one of the tested versions.

Thanks a lot for developing the firmware, it is so much work to maintain.

73 Frank

anigwei, 2023-03-06:
Hi @FrankUlbrich,

I just tried to send an APRS beacon with my Yaesu HT and receive it with portapack: It is properly received.

Best regards.

[image]

FrankUlbrich, 2023-03-08:
Hello anigwei,

thank you very much for your testing. Now I know, that it should work and the APP is not corrupted. 
Unfortunately no data is decoded , I will give it a try near a APRS repeater, maybe the modulation is more powerful.
I guess this bug report can be closed, because it works in anigwei s case.
Greetings to Espania

73
Frank / DO2FU

closed 2023-03-08 by FrankUlbrich