portapack-mayhem/mayhem-firmware #816: Radio not retuning when switching between apps

heurist1, 2023-03-03:
Tested Nightly release - 2023-03-02

Switching between applications shows the correct frequency on the GUI, but doesn't retune the receiver.

**To Reproduce**
Select Audio App.  Tune to a local radio station
Select Pocsag app, and turn on audio.
Local radio station heard demodulated NFM

Works the other way too
In Pocasg tune to a pager channel
Switch to Audio, and the GUI shows the last tuned signal, but the spectrogram shows the  pager channel.

**Expected behavior**
Retune occurs on app switch

**Affected versions**
* Latest Nightly release:

heurist1, 2023-03-03:
I have looked at the code, and I think it matches with the effect I am seeing.  Please shout if I have missed something obvious (or un obvious).  I have checked in the base class "View" to see if there was something hiding in there.

The frequency field in the GUI is set from the settings, but before the on_change callback is attached.
Or from the receiver_model if it isn't in the settings.

The receiver is updated using receiver_model.set_tuning_frequency(f), which is generally called by a method passed into the on_change and nowhere else. So the constructor will not setup the radio frequency.

This would appear to be true for other apps also.

Is there some magic operation that I am missing?  Does on change get called when it is first registered?

gullradriel, 2023-03-03:
The fact that some app do not change the tuned modulation is wanted, like if you use the scanner or recon app, you click on audio while on a frequency, then you're going into the audio app at that frequency.
Dedicated apps tend to set the frequency they use themselves, else they leave the choice to the user.

gullradriel, 2023-03-03:
Talking about POCSAG, when I see the wikipedia page (https://en.wikipedia.org/wiki/Radio-paging_code_No._1#National_implementations) it looks like it has a good lot of static frequencies and ranges.
The the perfect way to use that would be to first make one or more pocsag_band.txt freqman files, the using the scanner/recon app to loop through, and when you have a match going to pocsag. 
Then you'll be happy it kept the frequency in memory :-)
"Switch to Audio, and the GUI shows the last tuned signal, but the spectrogram shows the pager channel."  => could not reproduce it on my H1 & H2

loewal, 2023-03-03:
If I may add:
Some time ago I posted an issue about Pocsag.
My Hachrf Portapack in combination with Sdrangel show hundreds of messages in an hour.
Standalone nothing or perhaps one!
Always had a bad feeling about the initialisation of the Pocsag-app.
Could it be related?

Regards.

heurist1, 2023-03-05:
Hi [gullradriel](https://github.com/gullradriel)
_`"Switch to Audio, and the GUI shows the last tuned signal, but the spectrogram shows the pager channel." => could not reproduce it on my H1 & H2`_
I fully understand the functionality as you described and can see how that is very useful, but your description does not match either the code or my observations, but look at (link to code in Next)
[https://github.com/eried/portapack-mayhem/blob/3d248647c9020aa601184eb0a7e173dd64af9478/firmware/application/apps/analog_audio_app.cpp#L132](url)
analog_audio_app.cpp line 133->142.
It says that it will use the existing frequency if the settings failed to load, else it loads the settings.  
If it used the settings, then Line 138 puts the value in the GUI, but nothing loads it into the receiver_model.

Please could you humour me just a little more.
You didn't tell me what happened when you performed the test.  When you switch back from POCSAG to Audio App, did you get the POCSAG frequency or the previous Audio App frequency in the GUI, and which frequency was actually tuned/demodulated.
I can only see 2 options in the code.  1) You got the POCSAG frequency because your settings didn't load (if true please could you share your settings file from the SD card). 2) You got the previous Audio App frequency on the GUI but the radio was still tuned to POCSAG (which can't be a desired functionality).

gullradriel, 2023-03-05:
In the pocsag app I have the tuned audio app frequency showing up, no default prevailing. I'm not a pocsag app user so I don't know how it was working before.
Anyway whateve app I use that is tuning the frequency, I'm still on it when going in pocsag app.
Is that what you wanted me to confirm?

gullradriel, 2023-03-05:
When there is an on change event and you set the gui item it's calling the on change callback.
I'm on my phone actually and have no good access to a terminal, but will do tonight.

gullradriel, 2023-03-05:
I almost forgot to ask: what is your portapack version ?

heurist1, 2023-03-05:
I have a Chinese H2 (bought Dec 2020).  Never had a ny issues with it.

Yes agreed there is an on_change, which (in the audio_app) is set up after the field in the GUI is initialised, so it doesn't get called in the constructor.

I think I wanted you to 1) set a frequency in the audio 2) switch to pocasag and set a different frequency 3) switch back to audio, and tell me which frequency is displayed, and what is tuned to (pressing on the frequency to edit it will show you what is tuned, as apposed to what is displayed in the GUI).

Thank you for taking the time to help investigate this problem, it is very much appreciated.
and I know its a faff, but a copy of your settings files for the apps would be really useful.

gullradriel, 2023-03-05:
Seems you're right, something is broken / fishy.
I remember a time when it was working.

gullradriel, 2023-03-05:
If I tune in an app an go into another like pocsag, I see a freq that is not the one used in the previous one, but when I click on the button I effectively see the frequency that should have been tuned at start

gullradriel, 2023-03-05:
Ok, it's definitely coming from the 'load / save app settings' options in the 'Settings' menu. 
I get a correct behavior of the selected frequency when disabling them, can you test too please @heurist1  ?
More informations, I have a SD, a working battery button, and another unit with just the SD.
There is some inconstancy between SD app load/save and persistent memory usage / current settings used in receiver.

heurist1, 2023-03-05:
Hi @gullradriel.  
I have only one Portapack.  It has a working SD and backup battery.  I have tried turning off the load and save of settings, and it all appears to work as I would expect.  The effect is that the frequency doesn't get changed as I switch between apps, but the GUI is consistent with the tuned frequency, which is correct, and matches my expectations from reading the code.

The simplest (and no additional code) fix for this fault is to enable the on_change before loading the settings. That way we know that the radio will be consistent with the GUI.  I will make a proposed edit and raise a pull request for you to look, at, maybe tonight, or tomorrow evening.

gullradriel, 2023-03-05:
You are welcome, thanks for your work and contributions :-)
I'm waiting for it !

heurist1, 2023-03-06:
Fixed in PR #822

closed 2023-03-06 by heurist1