portapack-mayhem/mayhem-firmware #817: Otg cable

samukas81, 2023-03-03:
Last release 02/03/2023 does not recognize app (Android) via otg cable.

gullradriel, 2023-03-03:
Have you checked that your OTG is still working ? Have you checked the cable ?
Can't see where we touched something related to OTG.
Maybe a udev story.

samukas81, 2023-03-04:
Yes, cables all working, there are some apps on the playstore that I used as an sdr, until version 1.6.0 everything is ok.

jLynx, 2023-03-04:
Could yo try downgrade to your original version and see if that works?

samukas81, 2023-03-04:
v1.6 ok!

jLynx, 2023-03-04:
So it's working now?

samukas81, 2023-03-04:
yes because I downgraded to 1.6 again

jLynx, 2023-03-04:
Ah sorry, I misread your message, I thought you said everything was fine and 1.6 broke it but you mean after 1.6 it broke. I wonder if it's because of the HackRF firmware update @gullradriel?

gullradriel, 2023-03-04:
@jLynx maybe on android, if an app is restricting too much on the hardware id.
There is no such a problem using windows or linux, at least for me and @Brumi-2021 , with H1 & H2.
Maybe he should reach the apps author via the playstore/git/whatever and try to see with him.

gullradriel, 2023-03-21:
If there is really a problem, it's in the hackrf firmware and not in the portapack addon.
If there is still a problem for you, you should open an issue at hackrf's git: 
 https://github.com/greatscottgadgets/hackrf/issues.

closed 2023-03-21 by gullradriel