portapack-mayhem/mayhem-firmware #821: Pocsag again

loewal, 2023-03-05:
See attached file: Pocsag in SDRangel, recorded in 5 min.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to RECEIVE
2. Tap on POCSAG
3. WAIT....

**Expected behavior**
Several messages in, lets say, 5 minutes.
Result: nothing!
In SDRangel, same antenna, same frequency, etc: many!!!

**Affected versions**
I don't remember anymore, because its is not working for a long time now.

**Additional**
Please have a closer look.
[image]

heurist1, 2023-03-05:
Hi @loewal, no grantees, but I will try and have a look at this for you in the next week.  

I would require you to makes a "capture" of IQ data at 22KHz Rate and exactly the same gain settings as you use in the POCSAG app (that would be amp state and both gains).  I would also need to know what the gain settings were, what frequency you recorded, and what country you are in, and finally the build of software that you are running.  I wouldn't need a massive file, but would need at least one whole message block.  I think you should be able to see that on the capture spectrogram.  It might take you a couple of tries to get a good capture.

I have read the bit extractor in SDRAngel, and there is no reason why Portapack should be worse.  The Potapack uses a very similar algorithm, but also includes a mechanism to lock onto the message preamble regardless of the baud rate.  It even works on non standard rates.

loewal, 2023-03-06:
Hi @heurist1,

Thank you in advance!

However…which program should/can I use to record an IQ-file?
SDRangel or Portapack?

Please help and perhaps a step by step guidance..?

Regards

Loe

loewal, 2023-03-06:
> Hi @loewal,
> I think we should start with a capture on the Portpack, and then if that doesn't reveal anything, we can move onto looking at the output from SDRAngel.

Hi @heurist,

The problem gets more weird:

When I tune in at 172.450 MHz (Dutch Pocsag frequency) with the audio-app, I only hear white noise, not the well known pocsagnoise.
Tried 20 mtr., Air, Broadcast, all okay!

Switching modulation, e.g. from nfm to am gives a soundgap (silence), when a message comes in.

It looks like NFM is broken somehow.

I’ll try an different frequency with NFM.

…..later

heurist1, 2023-03-06:
There was a fault in reloading settings when entering the audio app, that got fixed over the weekend. So last nights build will be fixed. You can just do a quick check by editing the frequency after starting an app, if using an older image. Have a look at of the channel is not exactly on the frequency you expect. I have to run mine a few khz off expected to get it centered. It may be my LO is not as good as I expected.

loewal, 2023-03-07:
Tried to repeat what I've done...
Don't know which preset I selected in the looking glass...
Not working anymore....;-(

heurist1, 2023-03-07:
If you are able to try the latest nightly build, it would rule out the known retune problem. You may also have to delete your settings files (on SD card) for the apps, audio and postage.

loewal, 2023-03-07:
Noticed something else:
Setting the marker on some frequency and press Enter goes to the audio-app.

But there the frequency is the last one I selected myself and not the one from the looking glass.

heurist1, 2023-03-07:
Darn autocorrect. Yes POCSAG. I edited almost straight away, but you are getting messages by email, so I was too late.

Do you think you will be able to test with latest nightly build?  
The logic behind using an apps own frequency and using the last one set by the previous app is complex, and different users want different effects. It is also affected by the setting for loading and saving so settings. However if the audio is launched from within looking glass, we should probably make sure it forces the use of the currently tuned frequency. That is probably a bit too off topic for this issue. 

I am hoping to add a gui element to POCSAG so that the limits of the bit extraction will be visible, which will help with sorting out problems like yours.

loewal, 2023-03-07:
I’m already using that build.
What concerns me, is that even the audio isn’t visible. Except for once, when even Pocsag worked, coming from the audio app.

Don’t know what to do anymore. It’s quite frustrating that I had it working this morning (with latest build).

Loe Walter
Insula 103
6416 BZ  Heerlen
☎️ +31-6-24476963

loewal, 2023-03-08:
Working again...but not out of the box...:
- start looking glass
- select preset USA pagers 152
- set marker on 172 MHz
- select
- in Audio vary settings to both max en then to 8/24
- go to Pocsag

check settings: amp 1, 8/24

Works again!!

Is this logical?

gullradriel, 2023-03-08:
Yes ?
Pocsag is not tuning any frequency at start, it uses the last saved tuned frequency.
When you are going into looking glass you tune the frequency to the pocsag one.
Maybe trying to set the freq in pocsag would work too.

loewal, 2023-03-08:
I think it was just the gaincontrol.
I lowered it to 8/24. That’s all.
Now I don’t need the looking glass anymore.
Thanks!

gullradriel, 2023-03-08:
No problems.

closed 2023-03-08 by gullradriel