portapack-mayhem/mayhem-firmware #826: EPIRB decoder

Przemekmac1, 2023-03-08:
I think portapack could be great device to decode 406MHz transmition from EPIRB (emergency transmitters like ELT's, PLB's). 

https://www.sigidwiki.com/wiki/Distress_Radiobeacon_(Digital)
http://jgsenlis.free.fr/406.htm
https://www.youtube.com/watch?v=xLCC14b5wqU

Przemekmac1, 2023-03-09:
Hi.  It would be great to be able decode test signals by direct proximity of ELT, read country code, aircraft tail number and 15 HEX code.  
On french web its open C-code doing it.  I just compiled and it works well.  I can provide some WAV files recorded from aviation ELT's.

Przemekmac1, 2023-03-09:
[image]

[image]

Decoding by French software.

gullradriel, 2023-03-27:
I think it would be a good addition, like a survival/emergency/disaster mode X-D
Joke aside it would be interesting. @Brumi-2021 don't you think of any other existing app that is doing part of the decoding job ?

Przemekmac1, 2023-03-27:
Its no need to write it from the scratch.  I attached C code doing it. Here its next one, working on mobile platform:
https://github.com/djecom1/Sarduino-TTGO-T-Beam

Brumi-2021, 2023-03-27:
Hi @Przemekmac1 ,are you also available through discord ? 
I think it should be absolutely feasible to implement that app , and moreover with the reference code that you shared .

But just now I am busy with other app study . 
So I can not start now . Anyway if some one wants to start about it   , please go ahead and I will also try to cooperate. 

Anyway , I am convinced that its usage will be appreciated only  in lab only , because I feel very very very low probability to decode a real live signal  from UHF . 
But it is more a challenge ...

Cheers

Brumi-2021, 2023-03-27:
Join to discord Mayhem channel , just clicking the "chat" green icon in the main readme page , and contact me . 
Cheers

Przemekmac1, 2023-03-27:
Ok, I think I managed this.