portapack-mayhem/mayhem-firmware #828: Frequency counter

samukas81, 2023-03-10:
It would be interesting to have a frequency meter with the widest range possible.

gullradriel, 2023-03-10:
Something like the Calls app maybe ?

samukas81, 2023-03-10:
[image]

this

samukas81, 2023-03-10:
[image]

samukas81, 2023-03-10:
Unfortunately I don't know how to program, if I knew I would remove a lot of function that I think is of no use.  Example analog tv reception, pager and some others.  You have to leave the device clean and functional with real utility!  Example, rf generator with output control in dbm, spectrum analyzer similar to TinySA, frequency counter, wattmeter.

closed 2023-03-13 by gullradriel