portapack-mayhem/mayhem-firmware #842: Looking Glass marker doesn't work

loewal, 2023-03-21:
Hi,
At least in my Portapack H1, when I select the marker in the Looking Glass and press Enter, that frequency is not honoured!
Instead, it enter the Audio-app at the frequency that was last used in the Audio App or Pocsag...!

That worked before!

Running n_230321. Haven't tried older ones.

gullradriel, 2023-03-21:
We did not touch this.
It had to do with you enabling load and save app settings, or so I thinks.

gullradriel, 2023-03-21:
Tested it right now. It's a confirmed side effect of save/load app settings.
If you do not want frequency to be restored from last run, disable the save and load checkbox in your Settings/App Settings menu.

closed 2023-03-21 by gullradriel

loewal, 2023-03-21:
You’re right.
But it is not very logical, selecting a specific frequency and ending up in a saved one.

Regards

gullradriel, 2023-03-21:
That's logical if you enable per app settings saving. 
Maybe you can start a discussion about it on discord, and discard the frequency loading from certain apps.