portapack-mayhem/mayhem-firmware #851: PortaPack not working

MaxDeepCoder, 2023-03-24:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly verison here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

I cannot get the PortaPack H2 to start. The screen flashes momentarily and that is all.

I have tried many times to change the display driver with the buttons and tried everything.

The USB light doesn't go green.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the HackRF with the nightly release.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Nightly release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

https://user-images.githubusercontent.com/64033272/227491102-ee912798-dcf7-410b-a344-2327bfbb59bf.mov

gullradriel, 2023-03-24:
Where did you buy it ?
Was it working before previous nightly?
What is the last known working version on your device ?

maietta, 2023-03-25:
I guess it wasn't that urgent because OP got a reply in 2 hours but it's been 14 hours since.... so... not that urgent I guess.

The good thing is, if it's an official unit, it's designed that even if one think it is bricked, it's not. It can have the firmware flashed in one of two ways.

MaxDeepCoder, 2023-03-25:
No worry. I just put the stable release and it worked.

I wanted the microphone to work. 

The microphone meter is working but there is no roger beep or sound. Just a empty carrier.

Brumi-2021, 2023-03-25:
Hi @MaxDeepCoder are you available through discord ?
Did you find another stable fw release where it works the mic ? And How about other TX app ?

MaxDeepCoder, 2023-03-25:
Hi!

Yes I am available! 

Can I have your dc tag?

Brumi-2021, 2023-03-25:
Hello , I could not see your nickname in mayhem discord . 
Anyway I already gave you a good clue to try to fix your "no modulation, just carrier " , if you still have it .

MaxDeepCoder, 2023-03-25:
I read it and I saw that you need to take the PP and HF apart?

I can't do that because I am currently on a trip. Do you have any other solution

MaxDeepCoder, 2023-03-25:
Also, is there anyway to access the sd card from the portapack through USB? I have no SD readers. ☹️

jLynx, 2023-03-25:
> Also, is there anyway to access the sd card from the portapack through USB? I have no SD readers. ☹️

No that is not possible

MaxDeepCoder, 2023-03-25:
Ok.

JohnCanty, 2023-03-27:
I am writing to reply I have a similar issue. I purchased portapack H2 from Sigrid. I bought the HackRF One from Amazon (Noelec) and I installed the RF shield. After installing RF shield, i tested the hackRF as a stand alone SDR. I found my HackRF is one of the latest editions and some components changed, so it shows up as Board ID Number: 4 (An issue stated over in the HackRF firmware github). I have installed the latest stable release of Mayhem and once installed the hackRF board is pseudo bricked. I have to boot it in DFU mode for it to be recognized. The Portapack LCD flashes momentarily, but never boots, with or without the SD card installed. Any thoughts?

MaxDeepCoder, 2023-03-27:
Here is an idea. The SD card has the screen drivers and you have to change them by pressing and holding the knob to change screen drivers?

Correct me if I'm wrong

jLynx, 2023-03-27:
> Here is an idea. The SD card has the screen drivers and you have to change them by pressing and holding the knob to change screen drivers?
> 
> Correct me if I'm wrong

Correct

MaxDeepCoder, 2023-03-27:
> Are there more than the 4 lcd drivers available? Or is there a write up on
> identifying the screen and driver I need for this version I have?

Do you have a portapack h2+? Then use the left button

JohnCanty, 2023-03-28:
I have [this](https://www.amazon.com/dp/B0BRKH3JMB?psc=1&ref=ppx_yo2ov_dt_b_product_details). It is a Portapack H2+ from what I can tell. Also I don't see anything that obviously looks like an LCD driver on the files I copy to the SD card. Can you let me know where those files are?

MaxDeepCoder, 2023-03-28:
I have the exact same thing! It sounds weird but try and get into hackrf
mode without the screen? For some reason it put me into USB mode and not
dfu.

Another thing to try is to hold the left dpad button and click the knob to
power on, wait 10 seconds

MaxDeepCoder, 2023-03-28:
Also, I've tried many things to get the microphone feature to work and still no audio but it has a carrier. Could anyone help. I've read everything but its really confusing.

Brumi-2021, 2023-03-30:
Hi @MaxDeepCoder , did you solve it ? 
What is your real problem , Microphone base band not working, not moving vumeter ,or correct vumeter but just transmitting carrier without any modulation.  If it is that last case , did you tried to follow previous #466 ?

MaxDeepCoder, 2023-04-02:
Nope.
The VUMeter is moving but there is a carrier without any modulation.

MaxDeepCoder, 2023-04-02:
The sine/noise gen is not working and the soundboard is not working aswell. (as not working i mean no modulation)

Brumi-2021, 2023-04-02:
Then @MaxDeepCoder ,I suggest you to try to follow the  #466 .

Brumi-2021, 2023-04-08:
Hello @MaxDeepCoder ,  did you try to follow the steps described in #466 ? (Other members solved similar problem with it )

can we close that issue ?

closed 2023-05-10 by gullradriel