portapack-mayhem/mayhem-firmware #857: New Dir cannot select

loewal, 2023-03-29:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
H1 without touchscreen: cannot create FIRMWARE folder because cannot select "New Dir"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to File Manager
2. Try to reach "New Dir"...
3. 

**Expected behavior**
Want to create folder FIRMWARE in root of sd

**Affected versions**
All versions

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

loewal, 2023-03-29:
(temporary) Solution:
Create a file in the root of the sd-cars (e.g. by running Pocsag with LOG enabled).
Then I can use the scrollwheel to select that file and then got to New Dir.

So, without any single file in the root, It's not possible to create a New Dir.

loewal, 2023-03-29:
*s=d

gullradriel, 2023-03-29:
I confirm it exists if there is nothing on the sd card.

loewal, 2023-03-29:
If there are the standard folders then same issue. 
There must be at least 1 **FILE**

gullradriel, 2023-05-04:
Fixed in next / next stable release

closed 2023-05-04 by gullradriel