portapack-mayhem/mayhem-firmware #859: Is there a Windows desktop/laptop GUI to access and run portapack suite?

JasonBurchell76, 2023-03-29:
To whom it may concern,
First, everything everyone has done to make the HackRf and portapack is absolutely amazing. I love the features and functionality of the Porta-Pack system unfortunately, 30 years of wrenching on motorcycles and cars have left my hands pretty much useless. that being said, I have already destroyed 3 portapacks simply because I've lost the dexterity necessary to hold the device and interact with the touch screen GUI.
I'm hoping there's some way of utilizing the full functionality of the HackRf/portapack combo but through the nice ,big ,clunky laptop I have. If anyone has any thoughts or ideas or just knows where I can find what I described, I'd appreciate any help i can get. 

Thanks for your time. I look forward to the next latest and greatest collaboration from the GitHub Community.

Regards,
Jason Burchell
jasonburchell76@gmail.com
jlburchell76@yahoo.com

jLynx, 2023-04-10:
Hi @JasonBurchell76 this is currently not possible and there are no plans for it. Your best est would be using the HackRF by itself with GNU radio but you need a Linux based system to do that. Windows won't work (as far as I know). As for trying to use the touch screw, you should be able to use the device without touching the screen at all. My devices doesn't even have a touch screen. Or are you still not able to press the buttons. If so, what portapack version do you have?

AlphaGeek1, 2023-05-06:
@JasonBurchell76 try SDRangel.  It's not a set of utilities like Mayhem, but it can run (and works well for me) on Windows and it supports transmit for the HackRF One.
https://rgetz.github.io/sdrangel/index.html

There are lots of SDR programs for Windows that will work with HackRF One, Also, check out URF - Universal Radio Hacker which runs on Windows.  

I know this isn't exactly what you're hoping for.

jLynx, 2023-05-07:
Closing due to inactivity

closed 2023-05-07 by jLynx