portapack-mayhem/mayhem-firmware #861: Amplifier mode TX

samukas81, 2023-03-31:
Can anyone check if the amplifier is inverted in TX mode, where 0 = on and 1 = off.
For reception mode it is ok, 0 = off and 1 = on.

Thanks.

Brumi-2021, 2023-03-31:
Hi again @samukas81 ,
I found f those  chats , they were with Reza (in discord channel) . 
We both confirmed that current fw had good logic in RX AMP  using my Portapack or Sdrangel. But in his device the logic was reversed . He also tried to upgrade the Hackrf CPLD but we could not solve it . 

My confirmation plots

Brumi-2021, 2023-03-31:
Sorry accidentally closed , not my intention
Later I will attach some picturee

samukas81, 2023-03-31:
I discovered the defect, the amplifier circuit is burned, thanks for the answer.

Brumi-2021, 2023-03-31:
Hi @samukas81 , happy to see that you found the root cause 
Here you have my confirmation with sdrangel of my unit 
(all ok as expected , if you make zoom you can see the RX amp activated / de activated)
[image]

And here you have his confirmation of reversed functionality . Same observation in your case ? 
[image]

gullradriel, 2023-03-31:
I confirm it's working as expected on my H1 and H2.
We can close, but for anyone coming here: check if your amp is not burnt.

closed 2023-03-31 by gullradriel