portapack-mayhem/mayhem-firmware #867: SD storage

samukas81, 2023-04-03:
SD storage not recognize in windows 11 in function SD over USB.
[image]
[image]

loewal, 2023-04-03:
Install the driver that is supplied through the zip-file that also contains the binfile

bernd-herzog, 2023-04-03:
Thank you @richtoy this might be a very good hint.
We had some inconsitencies with this very new feature. Unfortunately I am unable to reproduce any error.

It should look like this when viewed "by connection"

[image]

samukas81, 2023-04-03:
Thank you very much to all!

closed 2023-04-03 by eried