portapack-mayhem/mayhem-firmware #872: Macro Button?!

helmarw, 2023-04-05:
is there a way to define you own buttons?
for example, if  i want to replay a certain file quickly: instead of going to replay, search and select the file and then press play
it would be nice if this could be done as user definable macro und link that to a button e.g on the transmit page in this case

There are certain ways how to go about it. recording actions while doing them on the device would be the user friendliest.
But probably easier would be to predefine an action button, e.g one in Receive and one in Transmit menu
and link that with a file on the SD card which will contain the commands it should do when pressed, which can be edited by the user...

eried, 2023-04-05:
not really, I mean you could alter the menu in the code easily but that's all

helmarw, 2023-04-05:
thought so, but would be a nice feature, or?!
like two custom buttons (one in receive and one in transmit i would say) where you can start any command via a file on the SD card.
similar to playlist (if it would work) but not only for transmit but also for record?

gullradriel, 2023-04-05:
Hmm maybe, and really maybe, I'd the lua experiment is a success, we can
imagine lua scripts driving the hackrf.

helmarw, 2023-04-06:
yes, that would be nice, if it works. that would give the user a lot more flexibility...

gullradriel, 2023-05-10:
Since the tests are done, not doable at least for now.

closed 2023-05-10 by gullradriel