portapack-mayhem/mayhem-firmware #873: TPMS app enhancement to support Celsius/Fahrenheit toggling

NotherNgineer, 2023-04-05:
I propose to enhance the TPMS app to support switching between Celsius & Fahrenheit temperature display, similar to the existing support for switching between kPA & PSI pressure.  The USA still uses the Fahrenheit temperature scale primarily.

For code simplification, I additionally propose to display "Pres & Temp" column headings in all cases, versus switching the column heading when units are changed.  Note that the units, kPA/PSI and C/F, will be displayed at the top of the screen already.

I have implemented and tested this change in the https://github.com/NotherNgineer/portapack-mayhem branch (filing this Enhancement request for tracking purposes).  Two files are affected by this change:  tpms_app.cpp, tpms_app.hpp

jLynx, 2023-04-05:
Can you please upload a screenshot of how your version looks?

NotherNgineer, 2023-04-05:
[image]

Attached screen-shot with this change implemented.

jLynx, 2023-04-05:
Ah cool, that looks good. Please open a PR so I can do a code review

NotherNgineer, 2023-04-06:
Closing - pull request completed

closed 2023-04-06 by NotherNgineer