portapack-mayhem/mayhem-firmware #876: There are strange errors

wzsySRC, 2023-04-06:
After upgrading to the Nightly Release -2023-04-06 version, the P.MemMgmt function displays "! Err.creating pmrm flagfile!" The battery is charged, but after restarting, the system time is displayed as "2015-01-01". After resetting the system time, it is still "2015-01-01". I tried to update to version 1.6.0, but it shows the same issue. What I can guarantee is that my batteries are intact and fully charged. I don't know where the problem lies.

gullradriel, 2023-04-06:
Hello.
When you reset you loose current time. You have to set it back.
PMemSettings are only usefull if you do not have a coin battery installed. Note that I'm not talking about the big battery.
If you try to activate, it needs the SETTINGS directory.

jLynx, 2023-04-06:
If you have the small coin cell battery installed and you are still getting this issue. Please try flash back to your older firmware that was working and let us know if that is still working or not

gullradriel, 2023-04-06:
Anyway it the pmemsetting can not set the flag file it's not activated.

gullradriel, 2023-04-06:
I tested it away with my H1 & H2, with and without a coin battery, with and without a SD card.
Do you still have the problem ?

gullradriel, 2023-04-26:
Closing due to inactivity and not reproducible.