portapack-mayhem/mayhem-firmware #879: TPMS does not detect some Schrader Sensors (Dodge/Chrysler/Mercedes)

helmarw, 2023-04-06:
I think i posted it some time ago, but after updating to 1.6.0 i tried it again.
Still nothing from my sensors. I got a lot from other cars but not mine. So in general its working just not all.

I also recorded the BB of one of my sensors i have lying around, and could replay it.
Its beeing detected by rtl_433 on my mac using an rtl_sdr but nothing on the portapack.
see screenshot. I can even see that portapack TMPS RX is receiving some signal but nothing gest decoded
for some reason. Its a pretty standard and common sensor wildly used in most Chrsyler /Dodge and also in a lot of Mercedes
so i dont know whats happening here

who is responsible for the TPMS part ?
I would be happy if i can support getting this fixed, just dont know where to start and what the problem might be
If you need more data or different BB from other Sensors just let me know, i'v a couple of those left over.
I could even send them if that helps..

[Schrader_ID19A0480.zip](https://github.com/eried/portapack-mayhem/files/11170970/Schrader_ID19A0480.zip)
[image]
[Schrader_ID19A0480_take2.zip](https://github.com/eried/portapack-mayhem/files/11171114/Schrader_ID19A0480_take2.zip)

loewal, 2023-04-06:
Same for me here.

helmarw, 2023-04-06:
thnx, i thought im the only one

loewal, 2023-04-07:
Have to correct myself (a little):
The screen stays blank, but I saw that there was a tpms.txt created.
Its contents:

20230407080427  433920000 FSK 38400 19200 Schrader C1319CB9B59A34056ACA/8DBE9763DCB37F3A4A2F

So, is this a bug?

loewal, 2023-04-07:
I had to put my microsd in an adapter on my iMac to see the contents of the txt-file.
SD over usb doesn't work (yet) on Macos.
Would it be possible to add a "view" button next to delete etc.... to view the contents of txt-files?

helmarw, 2023-04-07:
Are you sure this is your sensor?
I also recognize many sensors (including Schrader) as I ride, but none of them are mine.
(Schrader has a lot of different sensors, not all are the same!).
To test this, I got the Miller Tool 9936 to trigger the TMPS sensors on the bench.
I can see the transmitted data after triggering with rtl_433 and a RTL-SDR
but nothing happens in the portapack, neither on the screen nor in tpms.txt

also i can record the transmitted BB with the portapack an replay it. Same there nothing shows on the portapack but on rtl_433 it does...

loewal, 2023-04-07:
Sorry, overlooked the real problem.
My sensors for temp, pressure and humidity don't appear.
I would like to be able to enter the frequency manually.

helmarw, 2023-04-07:
TPMS with Humidity ? thats a new one! which car/sensor are you using ?
can you record BB and post it here!?

PS: SD card via USB doesnt work with portapack, and wont ever afaik. Nothing to do with MacOS

gullradriel, 2023-04-07:
SD card via usb was just added a few days ago.
No one is responsible for an app, nor the TPMS one

gullradriel, 2023-04-07:
I highly encourage people to try to start hacking themselves.

loewal, 2023-04-07:
If I could, I would..certainly!
I am of the days of Cobol, Pl1, little assembler.
Objectoriented languages are like Chinese for a Frenchman 🤓

helmarw, 2023-04-07:
if i knew where to start i would have already, but im not even seeing any error msgs nor do i know how to get them.
Is there a debug option on the firmware? im mean not the Menue Debug, something like a logfile to see whats going on ?!
or how do i see whats going on when receiving a TPMS packet ?

helmarw, 2023-04-07:
found my original post from 2020, thats how far i got at that time. Just wasnt enough feedback to actually kinda test it and then life got in the way for me to further continue at that time. maybe now if someone is actually able to program it or point me in the right direction....

just in case there are more people interested in improving the TMPS display, here is more information
https://github.com/furrtek/portapack-havoc/issues/349

NotherNgineer, 2023-04-23:
I'd guess proc_tpms is not matching any of the expected preambles if nothing is being written to the tpms.txt log file.  From the rtl_433 debug output and C16 files in issue furrtek#349, the preamble bit pattern may be something like 11101010101011010.  The rtl_433 debug output is also saying it's a standard "Schrader TPMS", not the "Schrader TPMS EG53MA4", and as far as I can tell rtl_433 code may ignore the preamble bit pattern entirely and only check the OOK packet length (should be 68 bits) and CRC to determine it's a TPMS sensor.  (Just FYI, thought I'd take a look at this since I recently made other fixes to the TPMS code.)

helmarw, 2023-04-24:
that would be great, i can test it for you, no problem!
if you have the possibility to trigger the sensor manually, i can also send you one of my spare sensors.

helmarw, 2023-04-25:
i just did a testbuy on Amazon to try a cheap Trigger device:
https://www.amazon.de/dp/B08R5VLYPZ?psc=1&ref=ppx_yo2ov_dt_b_product_details
working just fine for less than 12euros

the TMPS-Sensor in question is this one:
https://www.pkwteile.de/schrader/9482285

but you can find them cheaper on ebay or aliexpress, though they might be fine when triggered but not working in an actual car! i found that out the hard way ;)

sorry the video-overlay from my phone is mirrored but you get the idea. The device ID is on the label 19A0480
the same reported by rtl_433
https://user-images.githubusercontent.com/10408121/234269969-6a01fb56-eec9-4326-b459-9f2026df8dd0.mp4

NotherNgineer, 2023-04-25:
That's funny, I have the exact same TPMS Trigger device from Amazon sitting on my desk.  (I can't help but wonder if there's any way to get hackrf to transmit at the low 125KHz to emulate this trigger device, e.g. perhaps software flipping the carrier on & off).

Anyways I usually just shake my old TPMS sensors to trigger them (or drop them on my desk a few times).  When I put one in a covered pot of boiling water (in sealed bags) to check for 100C temperature output, it also transmitted regularly, perhaps because of the temperature change or perhaps from the bubbles.  :-)

Would you mind running rtl_433 again with verbose logging enabled -vvv and then trigger the sensors two different ways to see how the output pattern differs?:  (1) output when using Trigger device, (2) output when shaking sensor.

When I get my test code ready, I might drive over to a tire shop where they have a bucket of old removed sensors.  It might be a few days, but I know you're patient since you originally reported this issue in 2020.  :-)

NotherNgineer, 2023-05-14:
Just FYI, I noticed that Honda CRV TPMS sensors aren't showing up either, so I guess it's hit or miss whether a vehicle is supported or not.  But, while I've been side-tracked on other code changes, I'm still thinking about how the TPMS code is going to handle the incoming data stream when there is no preamble to check (a preamble helps to filter out random noise).  Eventually I will get back to this one...  :-)