portapack-mayhem/mayhem-firmware #883: Bricked after latest upgrade

loewal, 2023-04-07:
**Describe the issue**

Compiled latest source: [Merge pull request](https://github.com/eried/portapack-mayhem/commit/0d7072ec5d06eb9b568cc11f28622a87f4a51732) https://github.com/eried/portapack-mayhem/pull/882 [from bernd-herzog/fix_hackrf_mode_2](https://github.com/eried/portapack-mayhem/commit/0d7072ec5d06eb9b568cc11f28622a87f4a51732)

**To Reproduce**

Steps to reproduce the behavior:
1. used the script I always use without problems on my Mac
2. after flashing I get a flashing rx-led and black screen
3. had to unbrick

**My Hardware**
Running H1 with mayhem_nightly_n_230407_FIRMWARE

**Can you try the upgrade with a different PC/Portapack/HackRF?**
Yes, unbricked and flashed 230407 again

gullradriel, 2023-04-07:
Can you try the bin I posted in test drive please ?

loewal, 2023-04-07:
Yes!

Works fine. Thanks again.

gullradriel, 2023-04-07:
There is something fishy if your own compilation broke.
Did you 'make  clean' before calling make ?

gullradriel, 2023-04-12:
Well. If ever you'll open a new issue :-)

closed 2023-04-12 by gullradriel