portapack-mayhem/mayhem-firmware #886: GPS sim is not stable

AlexanderSkvortsov, 2023-04-08:
I have a problem with the GPS sim option. Every time the simulation process restarts, I lose the satellites (and the fix) for ~9-11 seconds. 
The maximum file size for FAT32 (SD card format) is ~30 minutes at a frequency of 1250000. 
If I need to have a Fix state 12 hours, how can I solve this problem?
One solution is to enable ExtFAT for the SD card. In this case 9-11 sec. is not a problem.
The second is a solution to the simulation problem, for example, in the restart mode with some file offset, where the satellites have a fix.

eried, 2023-04-08:
Option 3) stream from computer?

gullradriel, 2023-04-08:
That's not an issue with the firmware / hardware.
While we can continue to talk about your problem, I'm closing the issue.

AlexanderSkvortsov, 2023-04-09:
Options 3 ...LOL, is  much more unstable , in any time the stream cancelled without understandable reasons.

AlexanderSkvortsov, 2023-04-09:
If the options I can't usage as is, needed to do something. I can't see any reasons aganst extFat for the fimware. As options not for all.

jLynx, 2023-04-09:
I think it's valid and exFat would be the solution. But it's not a priority to get this fixed. I will re open

AlexanderSkvortsov, 2023-04-11:
... well, about fix and priority...The "fix" in this case is the set to 1 of _FS_EXFAT and some additional options. This "future" is already ready ))) in the source code.
I suppose 1-2 minutes for collaborator...

gullradriel, 2023-04-11:
I've just tried to do it.
While it's building with a few more warnings (fixable) it's failing because of the additional required size.
In shorter words, it does not fit anymore in the flash. 
So it will not be as easy as setting _FS_EXFAT to 1 ( and _USE_LFN is already set at 2 so nothing more is needed to activate support except for space....)

jLynx, 2023-04-11:
> ... well, about fix and priority...The "fix" in this case is the set to 1 of _FS_EXFAT and some additional options. This "future" is already ready ))) in the source code. Approximatelly 1-2 minutes for collaborator... I suppose.

If you can get it working, feel free to open a PR and I can review. But sounds like it needs a bit more work. Would be a great addition

AlexanderSkvortsov, 2023-04-11:
> I've just tried to do it. While it's building with a few more warnings (fixable) it's failing because of the additional required size. In shorter words, it does not fit anymore in the flash. So it [...]

In shorter words ... What does it mean?  Imppossible or possible or "hightly likely"? ) May be possible to cut some  dubious options, tools for example.

gullradriel, 2023-04-11:
But yes. If we can reduce the baseband code a bit it may be possible (it's yet to test)

AlexanderSkvortsov, 2023-04-12:
I can try, but I never done it. Do you have some instructions for this? I haven't time for waiting.

jLynx, 2023-04-12:
Everything is in the wiki for how to get started for development

gullradriel, 2023-04-12:
Just check the wiki on top of the bar.
There are tons of sections, you may use the scroll button on your mouse.
I prefer to not answer the last part of your message, since it would contribute to show how disgraceful that was.
Regards,
Gull

jLynx, 2023-04-12:
@gullradriel I don't think he was trying to be rude, I think he is saying he doesn't have the time to wait for one of us to do it, so he wants to do it himself

AlexanderSkvortsov, 2023-04-13:
> Just check the wiki on top of the bar. There are tons of sections, you may use the scroll button on your mouse. I prefer to not answer the last part of your message, since it would contribute to show how disgraceful that was. Regards, Gull

In Russian  we have the following thing

gullradriel, 2023-04-14:
Let's say I did not understand. You will still have to find time to install the environnement of your choice by following one of the guides here : https://github.com/eried/portapack-mayhem/wiki/Compile-firmware
Personally I use Debian.
Then you can use the few bits here for a basic: https://github.com/eried/portapack-mayhem/wiki/Software-Dev-Guides
You also have to be familiar with git and branches. After what you can start to dig in the code.
For what you want you will have to try to reduce code in the baseband source files. Something like 168 missing bytes. 
That or you can open the gps sim app code and see if you can mod it for your needs.
Be strong and good luck !

AlexanderSkvortsov, 2023-04-15:
Two days... I've been trying to compile a fork in Windows for two days. After a few hours, I have the next one every time:

5 10351.7 Ign:32 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse all Packages
#5 10471.8 Err:33 http://archive.ubuntu.com/ubuntu xenial-backports/main amd64 Packages
#5 10471.8   Connection timed out [IP: 185.125.190.39 80]
#5 10472.0 Ign:34 http://archive.ubuntu.com/ubuntu xenial-backports/main all Packages
#5 10592.0 Ign:35 http://archive.ubuntu.com/ubuntu xenial-backports/restricted amd64 Packages
#5 10592.3 Ign:36 http://archive.ubuntu.com/ubuntu xenial-backports/restricted all Packages
#5 10712.4 Ign:37 http://archive.ubuntu.com/ubuntu xenial-backports/universe amd64 Packages
#5 10712.7 Ign:38 http://archive.ubuntu.com/ubuntu xenial-backports/universe all Packages
#5 10832.7 Ign:39 http://archive.ubuntu.com/ubuntu xenial-backports/multiverse amd64 Packages
#5 10832.9 Ign:40 http://archive.ubuntu.com/ubuntu xenial-backports/multiverse all Packages
#5 10832.9 Reading package lists...
#5 10832.9 W: The repository 'http://security.ubuntu.com/ubuntu xenial-security Release' does not have a Release file.
#5 10832.9 W: The repository 'http://archive.ubuntu.com/ubuntu xenial Release' does not have a Release file.
#5 10832.9 W: The repository 'http://archive.ubuntu.com/ubuntu xenial-updates Release' does not have a Release file.
#5 10832.9 W: The repository 'http://archive.ubuntu.com/ubuntu xenial-backports Release' does not have a Release file.
#5 10832.9 E: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/main/binary-amd64/Packages  Connection timed out [IP: 185.125.190.39 80]
#5 10832.9 E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial/main/binary-amd64/Packages  Connection timed out [IP: 185.125.190.39 80]
#5 10832.9 E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-updates/main/binary-amd64/Packages  Connection timed out [IP: 185.125.190.39 80]
#5 10832.9 E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-backports/main/binary-amd64/Packages  Connection timed out [IP: 185.125.190.39 80]
#5 10832.9 E: Some index files failed to download. They have been ignored, or old ones used instead.
------
executor failed running [/bin/sh -c apt-get update &&   apt-get install -y git tar wget dfu-util cmake python3 ccache bzip2 liblz4-tool curl &&         apt-get -qy autoremove]: exit code: 100

This is not right place for my question maybe, but I'm not find the other.

jLynx, 2023-04-15:
If you are using Windows, please do it via docker and it will just work

AlexanderSkvortsov, 2023-04-15:
Well I was installing Debian and have the same

[ 10%] Building C object firmware/baseband/CMakeFiles/baseband_flash_utility.elf.dir/__/chibios-portapack/ext/fatfs/src/option/unicode.c.obj
[ 10%] Linking CXX executable baseband_flash_utility.elf
/opt/build/armbin/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: baseband_flash_utility.elf section `.text' will not fit in region `flash'
/opt/build/armbin/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 168 bytes
collect2: error: ld returned 1 exit status

Time to optimization...

jLynx, 2023-04-15:
Ah sorry I misread your error. Yes that's what @gullradriel was talking about earlier

> It means I submitted the code and error in code channel on discord. We are 168 bytes short on the baseband bin. For now I have other thing I want to finish, maybe someone will try his luck.

AlexanderSkvortsov, 2023-04-17:
There are two ways how resolve this problem.
Need to increase the size of the area for baseband_flash_utility (I haven't found it yet, this is my first expetience with such project) or reduce the size of the code by disable debug.cpp (as an example).

AlexanderSkvortsov, 2023-04-17:
The third way is optimize the debug text size.  I you calculate all 00 in the red rectanle you 'll get more than 168 bytes.

AlexanderSkvortsov, 2023-04-18:
But really, no needed to change the code.
If you'll change -O3 to -O2 you have the same .

#set(USE_OPT "-O3 -g -falign-functions=16 -fno-math-errno --specs=nano.specs")
to
set(USE_OPT "-O2 -g -falign-functions=16 -fno-math-errno --specs=nano.specs")

But I don't uderstand how it's possible?
...not all optimizations are equally useful!

bernd-herzog, 2023-04-21:
The blinking RX LED is indicating a crash in the application firmware.
See debug.cpp:113
`    if (source == CORTEX_M0)
        runtime_error(hackrf::one::led_rx);
`

AlexanderSkvortsov, 2023-04-21:
Well, I have finded the problem...
The _USE_LFN	was 2 .  I have changed it to 3 and device start witout any problems.  The new SD card 128 G with exFat format available too.

AlexanderSkvortsov, 2023-04-21:
In any case the sources  in the actual branch (next)  will have a compile problem.  The compiler uses 16 bytes for a 3-byte string. 
If you have 10 strings, you have lost 120 bytes as you can see in the red rectangle. 

One solution is to create an asm file and put all debugging lines into it. It may not be convenient for programming, but the size of the code will be reduced.

AlexanderSkvortsov, 2023-04-27:
Additionally... if anyone, ever adds exFat to the next bridge, please fix the following problem: File size > 4Gb in the list of folders on the device is incorrect. Devided by 4Gb now I guess.

Mainstreamlos, 2023-04-27:
Can anyone tell me which is the best/latest model of HackRF Portapack H2 Mayhem?

Mainstreamlos, 2023-04-27:
Is this model a good choice? 
HackRF Portapack H2+R4 Mayhem with CLPD: QFP100 and LTK8002D Audio amp and WM8731L QFN Codec.

AlexanderSkvortsov, 2023-04-27:
> Can anyone tell me which is the best/latest model of HackRF Portapack H2 Mayhem?

It depends on your application of this device. The latest sources you can download from the next branch.
https://github.com/eried/portapack-mayhem.git

The binaries from the main branch you can download here https://github.com/eried/portapack-mayhem/releases/tag/v1.6.0

Mainstreamlos, 2023-04-27:
> It depends on your application of this device. The latest sources you can download from the next branch. https://github.com/eried/portapack-mayhem.git
>
> The binaries from the main branch you can [...]

I mean the hardware mostly.
I saw this machine but there are not very many specs shown. Can you recommend this model?

PORTAPACK H2 0,5 ppm TCXO 3.2 "Touch LCD Display Black Aluminium, 2000mAh Battery, Speaker for HACKRF One SDR
https://a.aliexpress.com/_EjwoDVp

Mainstreamlos, 2023-04-27:
Can you recommend that the first screenshot with the H2-R4 with the WM8731 QFN Codec and CLPD QFP100 is the latest model, or do you recommend me another? 
I want Mayhem and download @Github stable & nightly firmware. 
In this case im a Newby and have to learn a lot, so sorry for my questions. Thank you for your help. 
I just don't want to buy twice or a wrong model. 
Just let me know if the H2-R4 version is the most compatibel and ready for future. 
Even let me know if another model would be better and why. What are the features? 

Regards Jan 

[image]
[image]
[image]

> It depends on your application of this device. The latest sources you can download from the next branch. https://github.com/eried/portapack-mayhem.git
>
> The binaries from the main branch you can [...]

[image]
[image]
[image]

Mainstreamlos, 2023-04-27:
Which model should I buy? 
I'm a little bit confused. 
For my understanding the H2-R4 version is the latest, but I heard I can't flash Mayhem 1.6.0. 
Only Nightly versions... 
Is it necessary? Which one is the best model for good compability and for flashing stable 1.6.0 & nightly Mayhem Versions? 
I'm new here, and the more I read the less I know. 
I only want the latest hardware and a model I can flash & update with the latest firmware. But which model is it?? 

[image]
[image]

closed 2023-04-28 by gullradriel