portapack-mayhem/mayhem-firmware #890: RX for Navtex Ship Weather Messages

parasender, 2023-04-10:
https://github.com/juerec/decoders

RX for Navtex Ship Weather Messages 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

blackwellj, 2023-09-09:
This would be great :)

Brumi-2021, 2025-05-24:
To @htotoo , if you check the above GitHub link, 
It is another clever way to demodulate fsk signals (navtext ship msgs , RTTY) ,I guess that  without fm demodulators, I guess that using directly the FFT of the  tuned raw data . 
(We already have FFT engine in the mayhem ) .

htotoo, 2025-05-27:
Can someone provide me some c16 samples please? Can't find any in the air with my current antennas :(