portapack-mayhem/mayhem-firmware #891: Amp mark

samukas81, 2023-04-10:
A suggestion, instead of marking 14 when turning on the amp, mark 10 because there are 10dbm of amplification.

Brumi-2021, 2023-04-10:
Hi @samukas81 , nothing is ideal , yes  the gain is not constant  is fully depending on frequency . But  by simplicity we are just displaying +14 dB's because it is the nominal value that mention the darasheet . I think is correct, no ?

samukas81, 2023-04-10:
Hello Brumi, it was just a suggestion because 14dbm is theoretical in transistor amplification, and 10dbm is found in more frequency than 14dbm.

Brumi-2021, 2023-04-10:
Hi @samukas81 , do not confuse about it . 

dbm : is related to output power unit related to 1mW (but we are not using this in Mayhem TX app menu absolute power, but it is true that nominal output power is 14 dbm’s). 

dB’s : is related to gain .In the TX GUI APP menus we are using relative chain gain (IF GAIN + AMP RF) , 
The gain goes from almost 14 dBs (HF area) to 10 dBs (6Ghz) . 

Strictly speaking , I agree with your point , but 
As  we are not talking to a professional calibrated product , I suggest not to make any sophisticated LUT (gain/frequency) to interpolate the real datasheet gain . I suggest  to keep that 14 dBs (as gain indication, knowing that it is only true < 0,1 Ghz) and that will fall down to 10 dBs at 6Ghz.  

your proposal to change to 10dBs , is going to the minimum gain (max freq ) while 14dBs is the Max gais (minimum freq) . 
 
  thanks for bringing it to an open discussion , and revisiting it , but. If no other strong opinion in next days , I  still suggest to leave as it is and close the issue. 

Cheers

closed 2023-04-11 by eried