portapack-mayhem/mayhem-firmware #893: portapack freezes when opening Recieve / Audio      Nightly Release - 2023-04-12

miotislucifugis, 2023-04-12:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
trying to open the Audio App under Recieve and the device freezes everytime... using touchscreen or cursor to select.
Running [Nightly Release - 2023-04-12](https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-04-12)
Portapack H2 w/ cpld100    
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Recieve'
2. Tap on 'Audio'
3. portapack freezes and requires a reset or power cycle

**Expected behavior**
Should ope audio app!

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:   [Nightly Release - 2023-04-12](https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-04-12)
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

maietta, 2023-04-12:
I noticed of my battery is low, I run into freezing much more often. Is this persistent or intermittent?

Haven't tried the current nightly built yet.

gullradriel, 2023-04-12:
Please check https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting

Particularly https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting#an-application-is-not-starting-anymore 

In your case go into SETTINGS folder and delete rx_audio.ini

gullradriel, 2023-04-12:
> I noticed of my battery is low, I run into freezing much more often. Is this persistent or intermittent?
> 
> Haven't tried the current nightly built yet.

I think your battery is just gently and lowly dying. But you should open up your own issue.

miotislucifugis, 2023-04-12:
Bingo-  sorry for the false alarm.    Great work on the updates btw!

gullradriel, 2023-04-12:
Nice to see it was it.
Thank you.