portapack-mayhem/mayhem-firmware #898: power module to portapack

ferdias07, 2023-04-13:
good afternoon
When connected to a computer or a power bank, the problem of interference in the portapack is generated in the battery charger of the portapak itself. When connected to the computer, or the powerbank,  and having the signal visible on the portapack, I have many interfering and harmonic signals.
I solved the problem by overriding battery charging IC from portapack and putting a battery charging module with IC tp4056, with battery overcharge protection(4.2V), battery discharge protection(2.9V) and with battery charge current control, just changing a resistance in the module.
With this battery charging module, even using a powerbank, I don't have any interference.

[image]

[image]

[image]

[image]

tankyou

gullradriel, 2023-04-13:
Thanks for the information. You should make it a wiki entry, that would be better than an issue since we do not build the portapack and you provided a solution.

euquiq, 2023-04-13:
I've been concerned about this situation since the birth of the first internal battery capable portapacks. Another hack that will solve this problem is -perhaps- adding a physical switch to connect / disconnect the charging module ? Sounds like a cumbersome "fix" but then it would give you 100% certainty in all cases / charging circuits that may come from those chinese labs.

eried, 2023-04-14:
Strange that I dont experience interference on mine while is charging, maybe they have been removing components for the filtering of the charger? (i.e. caps). But I also think it would be good to have this info as a wiki entry. Maybe a subsection of: https://github.com/eried/portapack-mayhem/wiki/Powering-the-PortaPack ?

Commander-Crash, 2023-04-16:
Im going to try this. Every portapack i had did this drives me nuts.

gullradriel, 2023-04-26:
As I'm linking the issue to the wiki and that it's already providing a solution, I'm closing it.
@Commander-Crash if you try give us feedback here please :-)

closed 2023-04-26 by gullradriel

pedroCX486, 2023-05-21:
Huh, I actually just found out about this issue and I'm also having issues. Got the Portapack recommended on the readme.md - mine actually has a problem where I can hear the charging led (the blue one that turns on and off) in a bunch of frequencies doing a loud beeping. I've been considering removing the leds (and losing battery indicator) because of that. I don't know yet if the charging IC is causing some other craziness like this, but will do some tests.

pedroCX486, 2023-05-25:
@ferdias07 Removing the IC has my Portapack H2 not turn on. Did you remove or just override it? Cheers.

ferdias07, 2023-05-25:
I not remove Theo power IC, i disconect  The + PIN.

ferdias07, 2023-05-25:
Thaís PIN , is The + PIN of The IC.

pedroCX486, 2023-05-25:
Ah, I see! Thanks! (PS: I speak Portuguese, feel free to mail me at pedro [at] orangemayhem.net, I want to write a blog post about this fix.

pedroCX486, 2023-05-25:
> Thaís PIN , is The + PIN of The IC. A quinta, 25/05/2023, 07:19, Fernando Martins ***@***.***> escreveu:
> […](#)
> I not remove Theo power IC, i disconect The + PIN. 

Ah, I see! Thanks! (PS: I speak Portuguese, feel free to mail me at pedro [at] orangemayhem.net, I want to write a blog post about this fix.

skiphansen, 2023-12-28:
@ferdias07 Could you tell me what the original charger IC is?  I just got my Portapack for Christmas and am trying to understand the behavior of the 4 battery status LEDS under the rotary switch.  I can't seem to charge mine reliably and I haven't been able to find any good description of what the LEDS mean.

pedroCX486, 2023-12-28:
@skiphansen The LEDs shows your battery percentages in 1/4 each, 0~25%, 25~50%, so on… fully lit is 100%. 

I have somewhere the schematic for the IC, I can upload later when I’m at my machine.

ferdias07, 2023-12-28:
my portapack have this IC:https://aliot.com.ua/pdf/eta6002.pdf, and this have interference. But the other portapack had more IC´s, IP5306...and others

skiphansen, 2023-12-29:
@ferdias07 Thanks for the quick response!  

I found my problem: a bad USB cable (how embarrassing).  

I had one flashing LED which I now gather was the 25% LED trying to tell me my battery was dead.  Now that I have a good cable I know which LED is which and that flashing apparently means charging or battery too low.  

I had assumed that the charger IC drove the LEDs directly, but I that's wrong or mine has a different chip.  

I haven't worked up the nerve to open it yet!  It was so flaky with an almost dead battery I thought I might be sending it back.

pedroCX486, 2024-01-07:
@skiphansen If you still need it:

[image]

skiphansen, 2024-01-07:
@pedroCX486 Thank you!