portapack-mayhem/mayhem-firmware #900: Black screen after update and DFU required to get hack rf mode back

CasualArclamp, 2023-04-18:
I will attach some pictures after the post. when on the latest hackRF firmware mt portapack h2 works and shows a blue screen with what that radio is doing. after I update to mayhem. the screen goes black and won't be picked up by the computer and the screen hotkeys for the changed chip don't work. then I have to use DFU for Hack rf mode to then flash back to normal firmware. plz help

jLynx, 2023-04-18:
Please specify what version you were on an what version you update to. As you just said you were on the latest and then updated which is impossible

CasualArclamp, 2023-04-18:
lateest hack rf standard firmware and latest mayhem firmware as of now

CasualArclamp, 2023-04-18:
[image]
![Uploading PXL_20230418_033635748.jpg)
[image]

jLynx, 2023-04-18:
I think it's normal for the H2 for the screen to go black in HarkRF mode, but not normal for it not to be detected. Maybe someone else with a H2 can confirm this. 

And if so, we should add a "known issues" section in the wiki @gullradriel

CasualArclamp, 2023-04-18:
[image]
[image]
[image]

CasualArclamp, 2023-04-18:
[image]

jLynx, 2023-04-18:
Thanks for providing the photos

CasualArclamp, 2023-04-18:
no problem. the main thing is the screen works on latest OG firmware

CasualArclamp, 2023-04-18:
update. latest nightly fixes issue

jLynx, 2023-04-18:
Awesome, glad it's now solved

closed 2023-04-18 by jLynx