portapack-mayhem/mayhem-firmware #901: Problem with new HackRF Rev. r9?

Dr-InSide, 2023-04-18:
Hi. 

My hardware: HackRF r9 board + PortaPack H2 v3.21.
Mayhem Software: Nightly Release - 2023-04-18
GreatScott Firmware: Relase 2023.01.01.

When I try to flash the Mayhem hackrf_one_usb via SPI update or via DFU update the TX+RX+USB LEDs are blinking and nothing works. I can't flash portapack-h1_h2-mayhem because the HackRF is not recognized anymore.

If I flash the original hackrf_one_usb from GreatScottGadgets, the HackRF runs without problems. I can now also flash portapack-h1_h2-mayhem, but then the HackRF runs only in USB mode. I can control it via USB and on the display of the PortaPack I see the blue status screen with the current USB settings. When I disconnect the USB cable, however, I don't see the PortaPack GUI. It is then only the blue status screen without values to see.

Why do the LEDs are blinking when I flash the Mayhem hackrf_one_usb?
What is the reason? Is there a solution for this?

Best regards.

mr-darker, 2023-04-23:
This link explains the issue and how to fix it. You can thank mrpeenut24 from reddit

https://hackerwarehouse.tv/product-knowledgebase/portapack-h2/hackrf-one-r9-and-portapack-compatibility/

comradecowboy9, 2023-04-30:
I'm doing the steps listed in that link and I'm receiving an error when trying to flash the SPI
"Unsupported Board IDCompatibility test failed."

Any advice?

DarkOctane, 2023-05-01:
> I'm doing the steps listed in that link and I'm receiving an error when trying to flash the SPI
> "Unsupported Board IDCompatibility test failed."
> 
> Any advice?

Double check you're using the correct .bin file?

fungaga77, 2023-05-11:
> I'm doing the steps listed in that link and I'm receiving an error when trying to flash the SPI "Unsupported Board IDCompatibility test failed."
> 
> Any advice?

Hi,
Iḿ having the same exact problem, with the same set (Hackrf One r9 gsg and portapack h2 v3.21- chinese).
-
""-andre@andre-Satellite-R630:~/Transferências$ hackrf_spiflash -w portapack-h1_h2-mayhem.bin
File size 992724 bytes.
Checking target device compatibility
Unknown Board IDCompatibility test failed.""

when i run hackrf_info-

""andre@andre-Satellite-R630:~$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc2a9560c3
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006f4f58 

Any news about this?

gullradriel, 2023-05-11:
Use -i in your command. Your hackrf_transfert is outdated.
Look:
hackrf_spiflash -h
Usage:
	-h, --help: this help
	-a, --address <n>: starting address (default: 0)
	-l, --length <n>: number of bytes to read (default: 1048576)
	-r, --read <filename>: Read data into file.
	-w, --write <filename>: Write data from file.
	**-i, --no-check: Skip check for firmware compatibility with target device.**
	-d, --device <serialnumber>: Serial number of device, if multiple devices
	-s, --status: Read SPI flash status registers before other operations.
	-c, --clear: Clear SPI flash status registers before other operations.
	-R, --reset: Reset HackRF after other operations.
	-v, --verbose: Verbose output.

fungaga77, 2023-05-11:
> Use -i in your command. Your hackrf_spiflash looks outdated.
> 
> hackrf_spiflash -h Usage: -h, --help: this help -a, --address : starting address (default: 0) -l, --length : number of bytes to read (default: 1048576) -r, --read : Read data into file. -w, --write : Write data from file. **-i, --no-check: Skip check for firmware compatibility with target device.** -d, --device : Serial number of device, if multiple devices -s, --status: Read SPI flash status registers before other operations. -c, --clear: Clear SPI flash status registers before other operations. -R, --reset: Reset HackRF after other operations. -v, --verbose: Verbose output.

Aparently , i just can read,reset or write...

""andre@andre-Satellite-R630:~/Documentos$ sudo hackrf_spiflash -i portapack-h1_h2-mayhem.bin
Specify either read, write, or reset option.
Usage:
	-h, --help: this help
	-a, --address <n>: starting address (default: 0)
	-l, --length <n>: number of bytes to read (default: 1048576)
	-r, --read <filename>: Read data into file.
	-w, --write <filename>: Write data from file.
	-i, --no-check: Skip check for firmware compatibility with target device.
	-d, --device <serialnumber>: Serial number of device, if multiple devices
	-s, --status: Read SPI flash status registers before other operations.
	-c, --clear: Clear SPI flash status registers before other operations.
	-R, --reset: Reset HackRF after other operations.
	-v, --verbose: Verbose output.
Thnks... im all new to Linux, code, git and hackrf...everything learned on line in the last 5 days, since received my set

gullradriel, 2023-05-11:
You're missing -w, -i was meant to be added not to replace

gullradriel, 2023-05-11:
Try this:
hackrf_spiflash -i -w portapack-h1_h2-mayhem.bin

fungaga77, 2023-05-11:
Like i said...noobie!! :)

Well...did that, unpluged it tried it on battery holding left button and...screen blinked nothing more :/ 

"andre@andre-Satellite-R630:~/Transferências/mayhem$ hackrf_spiflash -i -w portapack-h1_h2-mayhem.bin
File size 992980 bytes.
Erasing SPI flash.
Writing 992980 bytes at 0x000000.
andre@andre-Satellite-R630:~/Transferências/mayhem$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.

fungaga77, 2023-05-11:
Now, what happens is that i have to run DFU with hackrf-2018 and upgrade from there. If i try doing it hackrf-2023 direct , doesnt flash...
Is it a software or hardware problem?

gullradriel, 2023-05-12:
There are other closed/solved issues in the list about your problem, you may give the a read.

Brumi-2021, 2023-05-21:
Hello @Dr-InSide , let us  close that problem about fw upgrade to the hw HackRF R9 .  

We already covered it on #903 , and moreover , official fw version Mayhem 1.70 is already made based on latest HackRF compatible to HackRF R9 version .
So using Mayhem 1.70 or any later nightly, you should not have any problem . 

But if you still did not solve it , let us know and we will reopen it . 

Cheers

closed 2023-05-21 by Brumi-2021