portapack-mayhem/mayhem-firmware #903: Problem upgrading the firmware

mr-darker, 2023-04-20:
**Describe the issue**
My HackRF-One will not DFU-Update. It was bricked from trying to update mayhem from the factory provided version.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a bare-bones HackRF-One over USB.
2. Place it in DFU-Mode and attempt the batch-file method or linux dfu-util method of flashing the hackrf.
3. Observe 3 LEDs blinking, & that the device doesn't show up in device-manager, or lsusb unless placed back into dfu-mode.
4. Download the GreatScott HackRF Firmware & attempt the (same) linux-cli method with that .dfu file instead.
5. Note that the DFU image successfully loads, and the device is listed as HackRF.
6. Use the "hackrf_spiflash -w firmware-bin/hackrf_one_usb.bin" command to write to flash instead of just RAM.
7. Hit reset & observe a solid green USB light with the device detectable through device manager & lsusb as HackRF One.
8. Repeat the problem by going back to step 2 if desired.
9. Exercise poor judgement & try to continue with the PortaPak half of firmware update. It's bricked; back to step 1.

**My Hardware**
PortaPak H2+R4
HackRF-One Dated 9/11/22 purchased from the link below.
(https://www.aliexpress.us/item/2255800060726887.html?spm=a2g0o.order_list.order_list_main.5.21ef1802QkKZKK&gatewayAdapt=glo2usa&_randl_shipto=US)

**Affected versions**
I've tried version 1.6 as well as the nightly release.

**Were you able to update the firmware before?**
The firmware updates with GreatScott's only.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
I have a windows pc, and a linux laptop.

**Additional**
The three flashing LEDs are TX, RX and USB.

mr-darker, 2023-04-23:
In the instrucrions you gave, Step 5 turns into step 3 on my post.

The green light does not turn on after flashing, it starts blinking 3 LEDs instead.

It shows up in DFU mode on the PC, but does not appear at all after flashing. I have installed the drivers too.

mr-darker, 2023-04-23:
mrpeenut24 on reddit provided this link that explains the issue and how to fix it.

Its a compatibility issue with the R9 revision of HackRF

closed 2023-04-23 by mr-darker

gullradriel, 2023-04-23:
That link is good. Mr peanut (or ghost?) Should provide it as a wiki entry :-)