portapack-mayhem/mayhem-firmware #916: File Manager App , out of memory , when more than  aprox. 113 entries (files + directories)

Brumi-2021, 2023-04-24:
----
Confirmed that problem in latest fw 23rd April 23 nightly and many older versions (probably common to all version) 
 
----

**Describe the bug**
It is not a major problem.  Just  small problem , with some tentative solution .  
Now , In the recent fw versions, thanks to Bernd, when trying to access to File Manager , using my SD card  (that had many Captures C16 files , in my case , 145)  , I  got that hint .
Guru M0 Mediation "out of Memory" problem . and with front LED blinking

[image]

In the past from previous version ,at some point , I stated to have that  fw application crash , (and at that time , I alwasys thought that was my SD card with some error corruptions, but that was not correct) . 
 . 
The problem appears when 
1-Try to access  File Manager App with my current SD card that has 145 captures , and it keeps latch up , with that error.
2- Also possible to reproduce the same problem , with the Replay App. (also latch up , crash app)  

But there is no problem , using SD card , in the following App, 
1-) with the wav viewer App. (it works perfect) 
2-) nor with the Recording Capture App.(it works perfect) 

**Expected behavior**
see your tree folder structrue and  all the files that you have in the main folder , 
[image]

**Affected versions**
I believed that it is a Common problem from always, 

**Additional investigation **
With this example structure tree
[image]

The problem can not be reproduced  if the entries (files + folder) qty is  aprox  <= 109 
[image]

The problem can not be reproduced  if the entries (files + folder) qty is  aprox  > 113 
[image]

**Tentative solution**
Based on above test results,  
we have confirmed that the problem dissappear if you move the captured files (C16 + TXT)  to some subfolder , 
So if you want to avoid that problem , pls do not leave many captures to grow up in the main folder .... (better that you classify by areas or dates in subfolders) -

gullradriel, 2023-04-25:
It's a good search and I confirm the limitation.
We should put an explanation in the wiki for sure, and maybe mod the fileman so it's not crashing but telling the user that there is too much files /directory

gullradriel, 2023-04-25:
Anyway good job and thanks the task manager !

gullradriel, 2023-04-26:
We should really make dedicated directories for the LOGS, SCREENSHOTS, RECORDS
On top of that adding a limit from which we warn the user that there are too much files and that he have to clean

zxkmm, 2023-04-26:
Or if it is possible that don't spam all the file list into memory instead of push file name into memory for like firstly 100 files, and then if it reaches bottom (the highlight touched 100th file), then drop the old memory and load those after 100?

gullradriel, 2023-04-26:
A paged view with a button_with_encoder to move across pages would do the trick (scroll only on the displayed 100 entries, and use the page index to move by 100 lines blocks)

kallanreed, 2023-05-02:
I've spent a bit of time in this code and there are two sets of data. One is more easily paged than the other.
First the directory is enumerated into a vector of file entries. This would be the hard one to page while keeping a more or less consistent UI because the directory enumeration would have to be re-evaluated for each page, the list would be be rebuilt.

The other one is at the display layer. Assume the entries list has been fully populated, it would be easier to only show N items at a time in the menu control. Currently a new menu item is built for each entry in the entries list so we're basically duplicating the set. It should be reasonable to page the menu items.

Brumi-2021, 2023-05-02:
Hi @kallanreed , thanks for checking it .
I do not know if it would be possible to optimise the structure or architecture to handle much bigger entry number (folders + files) Than the 113 that I mentioned .
(I understand that you are considering split in display pages . That would be the optimum . 
But If it is too difficult to handle , I think that as a user , it would be maybe enough if we can send a GUI message when we detect (for example > 90 entries ) advising to the user  , something like that , “too many file entries in the main folder for our limited ram system , better that you consider to classify them in separate folders”. (Just my proposal ), 
Cheers

gullradriel, 2023-05-12:
@kallanreed have put in a 100 limit and a text message when the directories are full.
On top of that he also added nice cut/copy/paste to help organize the sdcard :-)

gullradriel, 2023-05-12:
I think we can close the issue, the 'paging' if it happens one day will be an enhancement.
Your call @Brumi-2021  ?

Brumi-2021, 2023-05-12:
Hi @kallanreed and @gullradriel  I  still did not try to compile the next . I will do this afternoon   but as long as you added a warning message to avoid that problem , we can close it 
And thanks @kallanreed for all the additional features and logos that you added . Really cool App , congratulations  🙏👍

closed 2023-05-12 by Brumi-2021