portapack-mayhem/mayhem-firmware #919: HackRF Mode after flashing in portapack h2+

Alexitmisty, 2023-04-26:
**HackRF Mode remains after flashing in portapack h2+**
I tried to download the latest stable version, the nightly version, and also modified version 1.4.0 of the firmware (removed the icons of several applications in ui_navigation.cpp). Everything ended with a "successful" download and freezing in hackrf mode. The solution is to rollback to jumbo77 version 1.4.2, but I need to learn how to upgrade the firmware. Could you please help solve the problem?

**To Reproduce**
1. Compile and flash with Using ARM on Debian
2. Download and flash last night version

**My Hardware**
H2+R4

**Affected versions**
Last nightly version, 1.6.0, modified jumbo77 1.4.0

**Were you able to update the firmware before?**
If installing version jumbo77 1.4.0 counts, then yes

**Can you try the upgrade with a different PC/Portapack/HackRF?**
I tried with one laptop on win11 and debian 11.x64bit in WMware. I used different wires.

[image]
[image]

zxkmm, 2023-04-26:
Pls close this as it already resolved.

Alexitmisty, 2023-04-26:
The solution turned out to be a flashing with pulling out the coin battery from the protapack.  Installed the latest nightly version.  After flashing, the start was made by pressing the power button and the right button.

closed 2023-04-26 by Alexitmisty