portapack-mayhem/mayhem-firmware #923: Signal generator

samukas81, 2023-04-27:
Can someone fix the signal generator, it only generates sine wave the others don't.

Brumi-2021, 2023-04-27:
Hello , I checked in mine ,  and all SG works (except noise that seems not working )
So we may need to have a look in the noise part only.

How about your other modulations ? if they are not working well in your device (FM , AM , USB . CW ...) 
, we already pointed the possible solution in other post .
 
CW ok , just carrier
[image]

SINE ok
[image]

TRIANGLE , ok
[image]

SAW UP. ok
[image]

SAW DOWN , ok
[image]

SQUARE , ok
[image]

NOISE , seems not working
[image]

samukas81, 2023-04-27:
Hello Brumi, I tested it on my Tektronix TDS-320 and the only way I can get it is sinusoidal, even in noise I have sinusoidal!

Brumi-2021, 2023-04-27:
Hi @samukas81 and how about the rest of TX modulations in your equipment  like  Mic  App,  or Morse App ?

samukas81, 2023-04-27:
Other functions are ok, it's been a while since I noticed this problem even in other firmware versions.  Is there no way to add other types of modulation in signal generator mode?  Another thing, would it be possible to change the frequency in the program itself, without having to leave it?

Brumi-2021, 2023-04-27:
Hello @samukas81 . before I checked with an H2 , WM audio IC ,  and now I reconfirmed with H1 , AK audio IC and exactly same results,   All modes works fine, except "noise generator" that appears just carrier .

Then I feel that your device has special problem . 
I suggest you to check two things, 

1-) You may have a very old CPLD  version on Hackrf board,  and producing those strange issues. , (see #651 ) , I had also that problem when I got the H1 . I could solve  it .

2-) Or you may have  some data corruption in the CPLD of the Hackrf,  if that is the case, the way to fix might be follow #466 -

Regarding your other consult ,  it is already possible to change the freq. inside the application , just move the cursor in below line and edit or apply rotation of the encoder . (but to change it , you have to be in NON TX MODE (means with ICON GREEN , "START" ) .

cheers.

samukas81, 2023-04-27:
Thank you Brumi, cheers!

Brumi-2021, 2023-04-28:
Hi @samukas81 , pls let us know if you could recognise any of the above problems and if you could fix it , rewriting again HackRF code first (that will update automatically the CPLD Iif needed ) , and Mayhem later 

by the way , we are working to fix the missing noise generator, hopefully coming soon ... 
(and then we will release a PR and close that issue) 

Cheers

samukas81, 2023-04-28:
Hello Brumi, I don't understand, do you want me to reinstall some other firmware?  I'm not a programmer otherwise I would try to help solve problems and requests from people.

Brumi-2021, 2023-04-29:
Hello @samukas81 , yes .
 If you have previous experience about reflashing tools and DFU proccess , and skills about dissasembling carefully the unit , my advice is dissasembling both boards Hackrf/ Portapack carefully , then pick up only Hackrf and apply two steps as I indicated above: (1) download latest Hackrf binary, reflash your Hackrf board with it  and confirm it communicates correctly to pc , hack_info , .... (2) after that , reflash it again with latest Mayhem nightly  . (3) re assembly both boards again, and check it ., and confirm if solved all those modulation issues.  (It is not to much complex , and if something goes wrong you always have DFU to recover it . But if you do not have skills or previous experience, better to ask some close friend help )
Cheers

samukas81, 2023-04-29:
Ok, I'll do it over the weekend and post the result.  Hug!

closed 2023-04-30 by gullradriel

Brumi-2021, 2023-04-30:
Hello, that Noise Generator  problem was found  last week , and now solved.   But looking its related  wiki entry , it seems it was already pointed in Feb  2021. 

[image]

Anyway I will update it accordingly today's PR .
Cheers