portapack-mayhem/mayhem-firmware #928: Recon Frequency field doesn't display frequencies correctly if there are leading zeros in floating part.

kallanreed, 2023-04-28:
```
	void ReconView::set_display_freq( int64_t freq )
	{
		int64_t freqMHz = ( freq / 1000000 );
		int64_t freqMhzFloatingPart = ( freq - ( 1000000 * freqMHz ) ) / 100 ;
		big_display.set( "FREQ: "+to_string_dec_int( freqMHz )+"."+to_string_dec_int( freqMhzFloatingPart )+" MHz" );
	}
```
This code does not display frequencies like 123.0075 correctly because the leading zeros in the fractional part are not preserved.
Would display like 123.75.

Use Recon and scan freqs that contain leading zeros in the fractional part.

Tested on head of next.

This seems like an edge case that's easy to fall into. Maybe a utility like `to_string_dec_int` would be good to have.

gullradriel, 2023-04-28:
You are right, it's a bug, and it should be around here from ages.
We are already having a  to_string_dec_int and that the problem is displaying it with a point to mark the MHz instead of displaying something like 125001000.
I'll provide a fix (for both Scanner & Recon) before the end of the week.

gullradriel, 2023-04-28:
Fixed, thanks.

closed 2023-04-28 by gullradriel

gullradriel, 2023-04-28:
For the records, it was to_string_short_freq ;-)