portapack-mayhem/mayhem-firmware #930: Bias tee function in hackrf-mode

Botox200, 2023-04-28:
Could it be possible to implement a function that starts the HACKRF-MODE with bias-tee switched on? 

This might make it possible to use the HACKRF-ONE with an active antenna for any program that has no bias-function for the HACKRF-ONE like SDR#-.

gullradriel, 2023-04-28:
In hackrf mode you pilot the hackrf from command line. Mayhem and the related firmware part are shut down.
To enable bias-t command line (before launching your tools, as they said):
hackrf_transfer -p 1
To disable bias-t from command line :
hackrf_transfer -p 0 
In hackrf command line, bias-t is named antenna_enable, antenna power.
Related command line help:
[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable.

gullradriel, 2023-04-28:
On top of that some tool are exposing it in their GUI's, with a name approaching bias-t or antenna power.
In that case, no need to use the cmd line, just tick the case in the GUI of the tool.

closed 2023-04-28 by gullradriel

Botox200, 2023-04-29:
Thanks, I'm familiar with the HRFO commandline commands. I am using this device mostly in Linux.

In Windows there is the extremely popular program called SDR# (sdrsharp) which does not support the bias mode for the HACKRF-ONE board. 

A command line activation of the bias tee in Windows conflicts with the SDR#. Thus is due to a  firmware upgrade of the HRFO.

So i am looking for a solution with a preset bias.

Thank you.