portapack-mayhem/mayhem-firmware #933: Jammer: Target SSID?

woofwoof, 2023-04-29:
Is it possible to target a specific WiFi SSID in Jammer? I imagine if a successful DOS attack takes place a sensible AP would try another channel. So, one could keep chasing the SSID and deny the appropriate channel as required?

Thanks for any feedback.

gullradriel, 2023-04-29:
It's not possible.

closed 2023-04-29 by gullradriel

woofwoof, 2023-05-17:
> It's not possible.

Why?

woofwoof, 2023-05-17:
Ok thanks. I imagine simply cycling through WiFi channels to detect an SSID. That's 13 discrete frequencies isn't it? Unsure of the speed of a router channel hopping to find a free frequency. Can't that be matched with the hackrf portapack? If not then could a e.g. raspberry pi or Jetson nano match a router using the hackrf over usb? Not sure half duplex is the problem. Though I see your point that full duplex would be better. I think polling though the over half duplex (receive/transmit) would solve that? I don't think a jammer signal would need to be continuous or example to be effective?

eried, 2023-05-18:
The jammer just generates noise, so it is not processing anything, maybe you could actually write something that tries to find patterns and only generate noise for that specific channel? I dont think targeting an specific ssid would be doable