portapack-mayhem/mayhem-firmware #943: Mayhem & HackRF rev (r9) TX woes

hacker671, 2023-05-02:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
When capturing signals for the replay app, signal is shown and captured, when it is then replayed only static can be heard from the receiving device. This also happens when using the soundboard transmission. I have only had 1 successful file played, since then even with same settings I get only static.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'replay/soundboard'
2. Tap on 'start'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.
Expected: sound to be played

Outcome: static only

**Affected versions**
5/01/23 nightly build

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-05-03:
Can you try using your reseller's firmware version ?

hacker671, 2023-05-03:
So I managed to discover something before I decided to re flash the firmware again, transmission works. But only if you do the following: go to transmit app, try to play a file(only static), go back to the main menu, and repeat this process 3 times.. on my third attempt to transmit I could hear the wav file or c16 file clear as day. I guess my question changes now to, why the hell?

gullradriel, 2023-05-03:
Which version are you using ?

hacker671, 2023-05-03:
> […](#)

Nightly build from 5/01/23, followed the wiki to a tee as far as install. Everything else works fine, it's just weird that I have to actively transmit then return to the main menu to hear only static the first 2 times. Only on the third does the file actually transmit.

gullradriel, 2023-05-03:
What settings are you using ? Sorry for the questions but it may be an hint.
I want to know:
Do you use a sdcard ?
If yes do you have the latest stable sdcard contents on it ?
Do you have a SETTINGS directory on the sdcard ?
Have you cleared it after reflash ?
Are you using app settings load and save ?
Do you have a coin battery installed?
Are you using the Pmem settings?
Thanks

gullradriel, 2023-05-03:
Also can you provide clear photos of both your hackrf and portapack board? The two sides of each. I know it's going to be tedious, but it can help us.

gullradriel, 2023-05-20:
Guess it is transmitting now since it's now 3 weeks without an answer.
Please reopen an issue or check the closed ones, come to discord to live debug your problem with available members.

closed 2023-05-20 by gullradriel