portapack-mayhem/mayhem-firmware #952: UI Navigation view stack doesn't work well with modal views.

kallanreed, 2023-05-03:
The problem happens when you try to write code that calls a view (like text_prompt) and then try to open a modal view in the callback.

The problem follows a sequence like this:
TextEntryView is put on the nav stack,
TextEntryView's handler is called
ModalView is put on the stack in handler
Handler returns and the nav stack is popped <-- BUG, this pops off the ModalView

Result: stuck on the TextEntryView.

I hacked around this by allowing a ModalView to be inserted "under" the current view so that the TextEntryView's call to pop affected the correct view. It appears there have been other hacks around this (see pop_modal).

One possible solution is to pop the nav stack _before_ the callback, but pop deletes the view object and so any state that is relied on from the view (like the on_change function pointer) is delete or being used after free.

Ideally:
- Every view would be pop itself off _before_ invoking its callback.
- This means we have solve the lifetime problem.
    - One way to do this would be to capture by value and to not allow capturing `this` in the callbacks. But that introduces possible memory concerns.

I'm open to ideas, but ideally this gets fixed across the board for all apps and then documented. 👍

kallanreed, 2023-05-10:
After looking a little more, I think it might be best to push the responsibility of popping to the caller. The tricky part is that most of the Views with callbacks don't always call back (e.g. ModalView with Yes/No only calls back for yes).
I think the best thing to do would be to only "nav.pop" in the View code if a callback is not defined. If a callback is defined, the View should expect the callback will call "nav.pop" instead.

gullradriel, 2023-05-10:
> I think the best thing to do would be to only "nav.pop" in the View code if a callback is not defined. If a callback is defined, the View should expect the callback will call "nav.pop" instead.
I would tend to go that way too