portapack-mayhem/mayhem-firmware #963: Add Pause to Playlist

ooooLoooo, 2023-05-08:
I really love the playlist feature and I use it a lot, but sometimes I need to add a pause for X seconds.

Would it be possible to add a pause feature so that I can add pause entries to the PLAYLIST.TXT?
Maybe something like this for a two minute pause:

315000000,SAMPLES/TeslaChargePort_US.C16,500000
pause 120
433920000,SAMPLES/TeslaChargePort_EU_AU.C16,500000

gullradriel, 2023-05-20:
That one would be simple to add and useful

ooooLoooo, 2023-06-09:
> That one would be simple to add and useful

I have been trying to give 10$ as a bounty (I know it is small money but I'm a poor man at the moment) to the one who adds this feature, but I can't get bountysource to work. I don't get the verification email they sent me, and I have contacted my email provider. Will put a bounty on this issue though, would love to give back to you guys, and will give more when I graduate and have a job :)

closed 2023-06-27 by ooooLoooo