portapack-mayhem/mayhem-firmware #965: H2 Encoder Steps.

ZxSpectruM55, 2023-05-09:
Hello, I follow your developments for portapack with interest and thank everyone who contributed.
My request will be about current encoder. My encoder is broken and I replaced it with a new one. but the old encoder was 30 steps. but I was able to find the one with 20 steps. I want you to add 30-20 options for the encoder to the settings section. I would be happy if possible.

gullradriel, 2023-05-20:
Hoooo that would explain why I had some skip on one of my devices.
Looks like a job for me.

ZxSpectruM55, 2023-05-21:
> Hoooo, bu, cihazlarımdan birinde neden bazı atlamalar yaptığımı açıklar. Benim için bir iş gibi görünüyor.

Thank you.

gullradriel, 2023-05-22:
I don't know for the number of encoding steps, but for my H1 & H2 this killed the bad 'reverse only after a few encoding delta' thing.

NotherNgineer, 2023-05-22:
AFAIK, I don't think current firmware has any idea of the number of encoder steps make a full rotation.  On a 20-step encoder, the user will just need to physically turn the dial 18-degrees per step versus 12-degrees.  :-)

ZxSpectruM55, 2023-05-22:
> H1 ve H2'mdeki sorunları kesinlikle düzelttim. @ZxSpectruM55En son kararlı test ettiniz mi? Peki ya kodlayıcı?

I tested the latest version. But the result was the same. The old original rotary encoder skips 30 steps in 1 full cycle. But I assembled the 20-step one. Frequency makes 2 jumps in one click.

gullradriel, 2023-05-22:
For that case maybe is there something like a step ratio or so ?
@NotherNgineer what do you think?

NotherNgineer, 2023-05-22:
@ZxSpectruM55:

Does this issue affect every app, or are you seeing it in a specific app?  For example, if you open a file in Notepad and use the rotary encoder to step through the characters, does it skip over every other character?

Secondly, it would be interesting to know what your see when you turn the dial very slowly in DEBUG -> BUTTON TEST.  Looking at the bottom two squares, we expect to see green-green -> blue-green -> blue-blue -> green-blue -> green-green, in that sequence when the knob is turned slowly clockwise.  If only one of the squares is turning green, there could be a loose wire.

NotherNgineer, 2023-05-23:
Inside the attached zip file, please find a bootleg firmware version with the rotary encoder dial sensitivity set to "low".  Please let us know if it resolves the issue.  Thanks!

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11547550/portapack-h1_h2-mayhem.bin.zip)

NotherNgineer, 2023-05-24:
I have revised the code, hence attaching a new bootleg firmware version with encoder sensitivity set to "low".  Please let us know if it resolves the issue.  Thanks!
[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11550548/portapack-h1_h2-mayhem.bin.zip)

ZxSpectruM55, 2023-05-24:
The problem is fixed. Now the rotary encoder is behaving normally.
Thank you!

> I have revised the code, hence attaching a new bootleg firmware version with encoder sensitivity set to "low". Please let us know if it resolves the issue. Thanks! [portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11550548/portapack-h1_h2-mayhem.bin.zip)

The problem is fixed. Now the rotary encoder is behaving normally.
Thank you!

NotherNgineer, 2023-05-26:
@ZxSpectruM55 
I have added a Settings page for the Encoder Dial, where the dial sensitivity can be set to Low, Normal, or High, and the setting is saved in persistent memory.  For your replacement encoder dial, you will need to change the sensitivity to "Low" and press "Save".  Can you please test this feature in the latest nightly firmware release, available at the link below?  Then this issue can be closed.  Thank you!

https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-26

ZxSpectruM55, 2023-05-26:
Yes, I tested it. It works perfectly. Thank you!

NotherNgineer, 2023-05-26:
@ZxSpectruM55
Thank you for testing it.  Can you please use the "Close as Completed" option at the bottom of this page?  (I don't have the ability to do that myself since I'm just a contributor.)

closed 2023-05-26 by ZxSpectruM55