portapack-mayhem/mayhem-firmware #986: Scanner App Enhancements (please assign to me)

NotherNgineer, 2023-05-14:
I'm working on the following usability improvements to the Scanner app and will submit a Pull request in the next couple of days.  Many of these were already implemented in the Recon app by a prior contributor.  If you have any comments or concerns, please let me know.

1.  The current Scanner app does not allow scrolling through the frequency list using the rotary encoder.  I'm changing the PAUSE/RESUME button to <PAUSE>/<RESUME>, and it will allow manual scrolling through frequencies (via the encoder) when focus is on this button. Frequency scrolling will be supported regardless whether or not in PAUSE mode (for forcing a skip to the next frequency before the Wait timer elapses).

2.  The current Wait timer forces a Skip to the next frequency when the timer elapses and a value of 0 means "never lock to a frequency" (which is not useful); I've changed the meaning of "0" to lock on the frequency forever (or until the signal is lost).

3.  Add a second configurable timer setting to maintain lock on a frequency for a specified number of seconds following signal loss.  This makes it easier to listen to a conversation on HAM radio (someone will often reply a second or two later) without having to press PAUSE.

4.  I propose to call these two timer fields "Wsa" (wait while signal active) and "Wsl" (wait after signal lost), versus the current "WAIT" field which is more ambiguous.

5.  I propose that the above "Timer" settings will be in 1-second units versus the current 0.5 second units, for easier understanding.  (The RECON app uses 1-ms units for two similar timers but I don't think it's necessary nor is there room on the screen)

6.  When a Freq file with a "Range" is read using the "LOAD" button, I propose to populate the Manual Start/End Search Range fields, versus trying to add up to 115 individual frequencies to the Freq List in memory (often overflowing this list meaning the whole range doesn't get searched).

7.  I propose to support Freq files which contain both Range and Single frequencies (but only a single Range in the file due to the change above).  Single frequencies will be added to the Freq List, and Range will be loaded into the Start/End Search fields.  Note that the current "ADD FQ" button already adds Single frequencies to a file with a Range in it.

8.  When loading from a Freq file, the default Modulation and Step and BW values shall be picked up from the file, if specified, versus leaving the settings on "AM, 9kHz step, 16kHz bw".  CAVEAT: At this time I am not (yet) planning to support a separate Modulation type for each frequency in the in-memory Freq List; the first valid value found is used.

9.  Support all BW and Step values in FreqMan (and use the table from FreqMan versus having separate tables in the Scanner code).

10.  Change the "Manual Scan" button to toggle between "SCAN Freq List" & "Sequential Range SEARCH" modes without having to re-LOAD the Freq file.  Keep Freq List in memory unless purposely cleared, and allow adding to this in-memory list while performing the Sequential Range Search.

11.  Add a CLEAR button to clear the Freq List in memory without having to load an empty Freq file.

jLynx, 2023-05-14:
It would be great if you could do each one of these as it's PR in order to keep each one small so there is not one massive PR to review if possible. Each one doesn't have to be it's own, but just not all together is possible since it woulds like it's gonna be a massive one. Also I have assigned to you now

NotherNgineer, 2023-05-14:
Good suggestion, and I'll think about how to split up the changes, if possible.  The good part is that I only need to update 2 files (ui_scanner.cpp & .hpp).

NotherNgineer, 2023-05-17:
I've added a screen-shot with call-outs to indicate what I've changed in the two Scanner source files, and I've attached it to the Pull request as well.  Sorry I was too far along on the code changes and I'd prefer not to have to submit each change as a separate pull request if at all possible.  I would however be glad to go through a code review with maintainers on a Zoom call if it would help.  Thank you for considering my suggested changes.

[image]

gullradriel, 2023-05-20:
@NotherNgineer are you done or are you converting the scanner to on_statistic_uodate only locking system?
:-)

NotherNgineer, 2023-05-20:
Sorry, forgot to close the ticket...  However, if you'd like me to fix or implement any other changes in Scanner then please let me know.

closed 2023-05-20 by NotherNgineer

dogger2448, 2025-12-12:
anyone know where the scanner app went im up to date on mayhem firmware and apps no scanner app and ai is to dumb to talk to so any help would be great