portapack-mayhem/mayhem-firmware #989: Buttons stop working intermittently (FW or HW issue TBD)

NotherNgineer, 2023-05-14:
Firmware stops responding to directional button presses intermittently, and the unresponsiveness lasts for several seconds at which point they begin working again.  Some observations:

1) Apps continues to run normally during this period, and still respond to rotary encoder & touchscreen access.

2) The Debug "Button Test" app shows no problems after hundreds of button presses.

3) In some cases, the buttons allow me to reposition to the upper left corner in an app, and press the center button to exit the app, but not in a right or downward direction.

4) I've seen this issue even on the home screen with no apps running (it's possible that I may have run and exited an app before this though).

5) I notice that buttons are (expectedly) unresponsive for a short period when performing file I/O to the SD card; might be a clue...
Speculation:  I'm wondering if SD card access (or other background activity) occurred and somehow failed to re-enable the buttons afterwards.  I'll be investigating along those lines...

Some people suspect a hardware issue, but I suspect it's a firmware issue, so I'm submitting this report for tracking purposes.  This affects FW version 1.7.0 on a (Chinese) Portapack H2+.

NotherNgineer, 2023-05-14:
Oddly, when the buttons become unresponsive, it seems that pushing the top & left buttons simultaneously resets the focus to the upper left corner & makes them all work again.  I haven't looked at the button handling code yet, but that might be a clue...

gullradriel, 2023-05-21:
I confirm the behavior on my H2 
And yes, left and right at the same time does 'unfreeze' in some case
Good luck in that quest @NotherNgineer  :-)

gullradriel, 2023-05-21:
To check the 5) I guess we just have to boot without a sdcard inserted and see if it freezes or not

NotherNgineer, 2023-05-22:
Please let me know if you still see the issue after PR #1030.

I do see that focus gets lost in the Search app if I move below the bottom of the screen (comes back if I press left button), but I suspect that's an issue in the specific app.

gullradriel, 2023-05-22:
Woooooo !
@NotherNgineer many thanks, flashed today and the stuttering is gone !
That is a great thing really :-)

NotherNgineer, 2023-05-22:
@gullradriel Thanks for confirming that my fix resolves the issue on your H2.  I'll close the ticket.  Looks like the debounce code goes back to the JBoone version, so I suspect every Portapack user has experienced those button issues.  I'd propose doing a 1.7.1 release soon.

closed 2023-05-22 by NotherNgineer

gullradriel, 2023-05-22:
That's improving both my H1 & H2 encoders: no more bad reverse when stepping freqs in recon !