portapack-mayhem/mayhem-firmware #1002: help me please

kewinxx, 2023-05-17:
----
I loaded a wrong firmware on a Chinese hackrf I removed the portapak so it doesn't work anymore and now when I connect it to the pc (windows 10) it doesn't recognize it and I can't do any operation the green light stays in the middle you have two keys( reset and dfu ) then steady red light in (rf) and flashing light (usb ,RX,TX)
if I connect it in (DFu) mode on the pc in the device management control panel it detects the item (LC) but I can't do anything how can I do???
please help me

Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/eried/portapack-mayhem/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-05-17:
Please first what was automatically added to your post, that is a lot of help already written down and even a video.

kewinxx, 2023-05-18:
gullradriel   thank you for the advice but from the device management settings it is written only LPC the writing hackerf one is not written

jLynx, 2023-05-18:
Closing this invalid issue it does not follow the issue template. Also your DFU driver is probsbly not installed correctly

closed 2023-05-18 by jLynx

gullradriel, 2023-05-19:
In the order, identify your version. Have a look at how it should be flashed. Use up to date tools. If it's still the shit, check "won't boot" and "update firmware troubleshooting". Don't forget to search for your problem in the closed issues, someone may already have it solved, mind you.

1: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions
2:  https://github.com/eried/portapack-mayhem/wiki/Update-firmware
3: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
4: https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting
5: https://github.com/eried/portapack-mayhem/issues?q=is%3Aissue+is%3Aclosed

Reset all settings to firmware factory defaults (only once boot to mayhem):
https://github.com/eried/portapack-mayhem/wiki/Factory-Defaults