portapack-mayhem/mayhem-firmware #1008: Out of memory

b4db0y01, 2023-05-18:
I turned on hackrf, he worked for 5 minutes and gave such an error, what should I do?
```
M0 Guru Meditation 
hind: Out of memory

```
Photo: https://cdn.discordapp.com/attachments/1106295501139230892/1108879267409952778/photo1684447568.jpeg

jLynx, 2023-05-18:
If you reboot by pressing the reset button. Does it happened again? Also what version are you on?

b4db0y01, 2023-05-18:
> If you reboot by pressing the reset button. Does it happened again? Also what version are you on?

After a reboot, the error reappeared. Firmware version 1.7.0

jLynx, 2023-05-18:
What portapack hardware version are you using?

b4db0y01, 2023-05-18:
H2

jLynx, 2023-05-18:
Need to know specifically what version as per the issue template you deleted https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

b4db0y01, 2023-05-18:
From OpenSourceSDRLab, HackRF One & PortaPack H2+R3

jLynx, 2023-05-18:
Awesome, thanks for that. I'll take a look and see if I can find anything

b4db0y01, 2023-05-18:
> Awesome, thanks for that. I'll take a look and see if I can find anything

Thank you!

b4db0y01, 2023-05-18:
Btw, I have another problem, it is that my hackrf does not save settings. For example: I go to AppSettings, check the boxes everywhere, but nothing is saved when the device is rebooted

jLynx, 2023-05-18:
Do you have the small coin cell Cr2032 battery installed and is it flat?

b4db0y01, 2023-05-18:
yes

b4db0y01, 2023-05-18:
> yes

I mean it's installed

b4db0y01, 2023-05-18:
but maybe the problem is that it is discharged

jLynx, 2023-05-18:
Check the voltage with a multimeter and/or swap it out and I bet your problem will be solved

gullradriel, 2023-05-18:
Check that you do not have too much files on the SD card too.

gullradriel, 2023-05-21:
In ADSB-RX there is a bug that is making it crash if there is too much planes. I dunno if it was here before.
What are you doing to be out of memory ?
We need to know the actions you made, in which apps please.
Also save and restore settings for apps is only for LVN/VGA/AMP/FREQ if I remember correctly.