portapack-mayhem/mayhem-firmware #1009: Painter sends mirrored text (and image)

loewal, 2023-05-19:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
All of my software (SDR++, SDRangel, Cubuicsdr) in combination with rspDX from SDRplay, receives a mirrored text.
Only when I check "Invert IQ" in SDR++, reception is okay.
Also checked this with a Galaxy tab S2 i.c.w. a RTLSDR. Same problem.
Cannopt understand why other people don't have this.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Transmit'
2. Tap on 'S.Painter'
3. select transmit text, "portapack"
4. gain:10, A:1, BW: 8,5, Du:10, P:5
5. run!

**Expected behavior**
"portapack" scrolling, but text is mirrored
Selecting "Invert IQ" and everything id okay.

I tried to explain in Discord, but was left somehow in the dark....

Brumi-2021, 2023-05-20:
Hi @loewal , as we discussed yesterday in Discord.
Normally any updated fw PP  device should behave like this , 
Means ,  using  PP Painter Spectrum App , selecting Offset Tuning (in SDR++side) , we should get correct Text when no selecting "Invert IQ" , and mirrored , when selecting "Invert IQ" (in SDR++ side) 

[image]

As your device was behaving opposite , we understand that clearly in your TX - RX path you had some invert IQ process.

Checking Hackrf github upstream , we eventually found , 
[image]

Then , as a first check , we asked you to update the Hackrf CPLD fw , and eventually you fixed it . 

In the past  , before 2021,  to update the Hackrf CPLD fw , we need to do it manually , with external command just to update the CPLD . But In latest  Hackrf releases,  (from Jan -2021 onwards) , it is much simple, 
then you can do it using old methode (direct) or much simple with two steps , 

1-) flash to Hackrf the latest  hackrf fw release.   (it will check automatically check CPLD version, and if needed it will be updated)
            hackrf_spiflash -w hackrf_one_usb.bin   (using current latest , Hackrf release, release 2023.01.1, or the latest) 

2-) On top reflash with latest Mayhem fw release. 
           hackrf_spiflash -w portapack-h1_h2-mayhem.bin (using current latest , Mayhem ex. 1.70 or any newer nightly .) 

I hope it helps to someone else, with same problem , 

If you agree @loewal , I want to close that issue . 

Cheers.

loewal, 2023-05-20:
Thank you @Brumi-2021,

One remark: there is no Offset Tuning visible when operating a SDRplay rspDX.
I am running the latest SDR++ by the way.

The recipe you describe with the 2 files is nice, but sometimes this doesn’t work and mirroring is back back again.
I haven’t found yet out why this happens.

Keep you and others informed….

Greets,

Tournesol

gullradriel, 2023-05-20:
The only way it can come back is if you flash something containing an older cpld.

closed 2024-01-05 by gullradriel