portapack-mayhem/mayhem-firmware #1017: ADS-B RX list is cleared out by time: setting or bug?

kw3-research, 2023-05-19:
**Describe the bug**
on**v1.7.0** 
ADS-B, RX, list of received planes on screen is cleared out after a certain amount of time (few minutes)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Receive '
2. Tap on 'ADS-B '
3. leave the device on and observe when an event is recorded, then removed

**Expected behavior**
This behaviour was not noticed on 1.5.4, list was maintained and it was possible to scroll it until the rx page remains active.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: _**1.7.0** ADBS-D RX list erased on screen in timely manner_
* Latest Nightly release: to be tested/checked
* Previous working release: _**1.5.4** ADS-B RX list maintained on screen (then written on file at page exit)_

Any tought is welcome,
Thanks

Cheers!

jLynx, 2023-05-19:
Maybe @heurist1 has some ideas?

kw3-research, 2023-05-20:
update 20-05-2023

Issue is still present with latest Nightly release - 2023-05-20
the station received disappear from the list when age counter reaches 300

**Age counter** actions on received station on list:

**0 to 9** -> station line remains white
**10 to 29** -> station line is light grey l
**30 to 299** -> station line is dark grey
**300** -> station canceled/removed from the list

gullradriel, 2023-05-21:
Hello, 
It's by design as stated by that line:
../firmware/application/apps/ui_adsb_rx.hpp:#define ADSB_REMOVE 300  // Used for removing old entries

gullradriel, 2023-05-21:
Wrong button. Didn't want to close before having your feedback.

kw3-research, 2023-05-21:
Hi,
thanks for that (I'll see if I'm able to edit this parameter and change according my expectations)
BTW it seems this was not set on 1.5.4, or if so, it was not working :-)

ZxSpectruM55, 2023-05-21:
Ads-b doesn't find any planes. I'm using the last nightly version.

eugene210492, 2023-05-21:
> Ads-b не находит никаких самолетов. Я использую последнюю версию nightly.
Try enabling AMP

ZxSpectruM55, 2023-05-23:
Already active. Ais is working. Probably the antenna is bad.

kw3-research, 2023-05-24:
relevant information to mitigate/fix the issue received, thanks.
../firmware/application/apps/ui_adsb_rx.hpp:#define ADSB_REMOVE 300 // Used for removing old entries

Will proceed to configure this parameter for my needs.

Cheers!

closed 2023-05-24 by kw3-research