portapack-mayhem/mayhem-firmware #1018: problem on pc window 10 in device manager only appears written (LPC) hackrf one disappeared how to restore I installed the drivers but nothing

kewinxx, 2023-05-20:
----
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

----

**Describe the issue**
A clear and concise description of what the issue you are facing is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'

**My Hardware**
Please specify what PortaPack hardware version you are using. 
You can find the list of versions here: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions

**Affected versions**
Please tell us what version you are running. 
If your device is still functional, try the latest nightly release before submitting this. 
You can find the latest nightly version here https://github.com/eried/portapack-mayhem/releases

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
If the issue is difficult to explain, additionally to the text please include images and videos.

jLynx, 2023-05-20:
Closed due to invalid issue

closed 2023-05-20 by jLynx

kewinxx, 2023-05-21:
> Chiuso per problema non valido
hello thank you very much for the help you gave me can you tell me where to post videos of my problem???

jLynx, 2023-05-21:
https://github.com/eried/portapack-mayhem/wiki/Diagnose-firmware-update-in-Windows