portapack-mayhem/mayhem-firmware #1051: Car fob key decode

utenus, 2023-05-23:
I guess it will be a very hard task and I'm trying to figure it out myself, but I think it's worth trying to add car key decoding functionality to PortaPack. Fantom Grabber has similar functions. It even has an official website. I think it can help in testing security and no one forces anyone to use it illegally, but I would like to check if my car is vulnerable.

[Pandora code grabbers](https://agentgrabber.com/en/product-category/code-grabbers/pandora-code-grabbers/)
[Crypto grabbers](https://agentgrabber.com/en/product-category/code-grabbers/crypto-grabber/)
[Preview](https://www.youtube.com/watch?v=sNqBJIvwzV4)

eried, 2023-05-23:
Hi, I understand that it would be cool to have all that for free, but without access to the algorithms, nobody here can add anything to the firmware.

If your car is on their official website, then is vulnerable.

closed 2023-05-23 by eried

utenus, 2023-05-26:
I understand, and it's a bad idea to leave a issue in the hope that someone will take up the topic? I wrote to the developers of these tools asking for support for the project. Maybe I'll set up a fundraiser for these devices and try to reverse engineer them. Or maybe we need to try to get these algorithms in different way.

eried, 2023-05-29:
I can be an open issue but realistically there wont be any development without: 1) access to code or research related with this or 2) an insane amount of money in a bounty -> https://github.com/eried/portapack-mayhem/#what-if-i-really-want-something-specific

If you find a solution, feel free to ping back. Keep in mind that you also need access to the hardware for 2) to be feasible, and 1) to be test-able, so there is many layers of obstacles.

hongo214, 2024-11-29:
[image]
I'm trying to figure out how to decide Manchester key fob.